genodelabs/genode #2683: vfs_lxip: skip internal Linux connect for UDP

ehmry, 2018-02-15:


ehmry, 2018-02-15:
UDP connect with Lxip is failing. Fixed at 5903961deb984a6f9145b996b272542ace81234e.

ehmry, 2018-02-17:
@nfeske A have a patch to quiet the output of lxip at 2ab850207c41f7587406338307e045f467e90526.

chelmuth, 2018-02-19:
Could you please elaborate a bit more why you propose to not inform the lxip stack about the connected UDP state? Doesn't this change the processing of incoming UDP packets that originate from not connected peers?

ehmry, 2018-02-19:
I suspect that the internal connect hook will always fail for UDP and that this case is instead handled in the Linux libc. But you are right to point out that connect should filter packets from unconnected peers, which would now not happen. I will look into what Linux is really doing.

ehmry, 2018-02-20:
The real problem is that UDP connect uses `ip4_datagram_connect`, which is currently a dummy stub.

Fixed at 2ca90e3d0a35596af2c700c67c417cc896e07afc, the previous patches for this issue should be reverted.

chelmuth, 2018-02-20:
Thanks for the fix!

commit 6cad6c13 (2018-02-19): Vfs_lxip: skip Linux connect for UDP sockets

The internal Linux connect operation is not appropriate for UDP sockets,
but setting the sockaddr metadata for the remote host internal to the
VFS plugin is sufficient.

Fix #2683
commit 6de469f3 (2018-02-20): dde_linux: Build net/ipv4/datagram.c for UDP connect

Fix #2683
commit 590f51db (2018-02-28): dde_linux: Build net/ipv4/datagram.c for UDP connect

Fix #2683
closed 2018-02-28 by chelmuth