genodelabs/genode #2686: Noux: rmdir dir/ (trailing slash) does not work

nfeske, 2018-02-20:
When using bash's tab completion on a directory, the completed name has a trailing slash. However, rmdir stumbles over an directory argument with a trailing slash and refuses to delete the specified directory. It works after manually removing the trailing slash though.

As a possible solution, the libc could strip trailing slashes of the rmdir argument. Maybe, a similar issue exists for other file/path operations?

NobodyIII, 2018-03-05:
This is probably related to "rm -r dir/" failing to delete the directory itself. I would love to see this fixed.

It sounds like our libc was designed for a less strict VFS implementation that allowed trailing slashes in paths. We probably need to remove trailing slashes from all libc file path strings in order to get the standard unix behavior.

chelmuth, 2018-03-08:
It seems the libc could expect the kernel to strip the slash as `strace rmdir /tmp/delme/` shows on Linux.
```
rmdir("/tmp/delme/")                    = 0
```
So, we have to strip trailing slashes in *libc/file_operations.cc* `rmdir()`.

nfeske, 2020-04-23:
Removed noux from the title. As far as I know, the problem still exists when using bash.

nfeske, 2020-05-04:
Commit https://github.com/genodelabs/genode/commit/eb0a33302a0dfc0bf24a59a48a84f86bf390ab0d finally addresses this issue.

commit eb0a3330 (2020-05-04): libc: strip trailing '/' for directory operations

Fixes #2686
closed 2020-05-18 by nfeske