genodelabs/genode #2687: Noux: gets stuck on attempt to exec a text file

nfeske, 2018-02-20:
When accidentally attempting to execute a configuration file in sculpt, noux rightfully complains that the to-be-executed file is not an ELF binary. However, it does not gracefully recover from this situation. So the user has to force a restart of the noux instance.

cproc, 2018-02-23:
The error message about the invalid ELF executable is printed by ```Child::_try_construct_env_dependent_members()``` and Noux does not get to know about the error. From its perspective, the 'execve()' call was successful and then 'bash' waits forever for the child process to exit.

Commit d28ca56 adds a simple ELF signature check in Noux, but perhaps it would be better to let the 'Child' report errors it encountered.

nfeske, 2018-02-23:
I agree. I very much appreciate your interim solution though. ;-)

BTW, noux already takes a look into the binary to handle shell scripts starting with "!#". This happens in `_process_binary_name_and_args`. I haven't inspected the code but may it be possible to move the check there? This would alleviate the need to attach the binary another time.

nfeske, 2018-03-01:
Fixed in the master branch.

closed 2018-03-01 by nfeske