genodelabs/genode #2691: rom_filter: proposed additions to handle some special cases

BorisMulder-CSL, 2018-02-26:
Hello,
I've made some scenarios where the current functions of rom_filter are not sufficient to reach my goal. Therefore, I have made some changes to the rom_filter. It would be nice if you could look into them and could consider adding them to the genode repository, or provide some criticism.

- First of all, an Input_rom_registry::Entry does not read from the ROM module upon initialization; instead it will always display an <empty/> node. Only when the input ROM session gives an update signal will it do so. This might give problems in cases where this input is immediately needed in the output and the ROM is not immediately updated (so the signal is not sent). I have made a minor change to make sure this ROM is immediately parsed upon constructing the Entry: https://github.com/nlcsl/genode/commit/1cd56c483f7e43dd742e2d03d34dc246f4530bc1

- Second, whenever the <input> node is used within the <output> node of rom_filter's config, an attribute called 'sub_node' has to be specified which determines that only the input ROM's sub nodes of that type will be included into the output. In practice, sometimes you want all the sub nodes of the input ROM to be copied. I added this functionality so that the sub_node attribute can be left out, in which case it will take all the sub nodes from the input. See https://github.com/nlcsl/genode/commit/6ec2c7ff71aee8bf760f71d055c609867c1eb093

- Third, the input ROMs configuration features a <node type="foo"> tag where it searches in the specified xml for a sub node called foo. However, if there are multiple such sub nodes, it will always pick the first sub node. Sometimes, you want it to select a specific sub node based on some attribute of that sub node. For example, in the following xml:
```
<xml>
  <foo name="bar">
    ...
  </foo>
  <foo name="baz">
    <myvar value="yes"/>
  </foo>
</xml>
```
  One might want to get the value of myvar. For this case I added a feature where the attribute of a node can be matched. to create an <input> node in the rom_filter's config based on this variable, you could do the following:
```
<input name="xml" rom="foobar" node="xml">
  <node type="foo" attribute_name="name" attribute_value="baz">
    <node type="myvar">
      <attribute name="value"/>
    </node>
  </node>
</input>
```
  I am not sure if the "attribute_name" or "attribute_value" attributes are the best names for this, so better suggestions are welcome. I added a commit which implements this feature here: https://github.com/nlcsl/genode/commit/bf9c893239addcb8c8f347570c551793d8f4742f.

- Finally, I changed the test script to include the latter two cases. It can be seen here: https://github.com/nlcsl/genode/commit/8c33dfaad868d79f1699c9f71af722cc6ff34806.

nfeske, 2018-04-05:
Thanks @BorisMulder-CSL for these proposals, patches, and the improved test. Now that the release stress settled, I had a chance to take a closer look.

All of your suggestions are reasonable. I took your patches along with a few fixups to my topic branch  (https://github.com/nfeske/genode/commits/rom_filter). Note that I shortened the first line of each commit message to stay within the limit of 65 characters (as a good practice). Regarding the naming, I changed `attribute_name` to `attribute`, and `attribute_value` to `value` as I see no risk of ambiguities here. In https://github.com/nfeske/genode/commit/9bd810c59c1b2b459fc59682935cf02341d2c05a I simplified/refined the condition checks. I sense that your original version seems to leave a few corner cases unaddressed. I hope that my change is in line with your intention.

The `sub_node` feature leaves a bad taste in my mouth. Not because of your extension but because it slipped in with https://github.com/cnuke/genode/commit/7263cae5a18b4f1f2293d031f9bafcf05ba51146 in the first place. Instead of selecting input at the `<output>` node, it should be selected by the `<input>` nodes, which would be consistent with the path traversal there. As of now, the selection of input happens at both places, which is bad.

@cnuke Can you remember the original motivation for this feature? Can we possibly satisfy its use cases with the `<input>` path selection feature instead?

BorisMulder-CSL, 2018-04-05:
Looks good.
I also agree that the sub_node functionality should be done by the `<input>` node instead.

nfeske, 2018-04-11:
I checked back with @cnuke and he agrees that the node selection at the output node can be removed. His original use case vanished. So I would add a commit for removing the code, which - however - nullifies the commit https://github.com/nlcsl/genode/commit/6ec2c7ff71aee8bf760f71d055c609867c1eb093.

@BorisMulder-CSL This means that the feature of inserting multiple subnodes won't be available unless we change the input-node handling in this respect.

nfeske, 2018-04-11:
I updated my branch https://github.com/nfeske/genode/commits/rom_filter accordingly.

@BorisMulder-CSL Since your code has been completely replaced by now, I replaced your commits by completely new ones. I hope that is ok for you. I covered the attribute matching by the test in a simpler - I think easier to understand - way and included this change in the feature commit.

BorisMulder-CSL, 2018-04-30:
@nfeske All in all, your version looks fine. There is a specific use case in my scenario though which broke because of the removal of support for inserting subnodes of a certain type:

I am using my rom_filter to generate a config of an init component. This init has a long list of start nodes. In some cases depending on <if> nodes in the rom_filter, I want all of these start nodes to be turned either on or off.
However, I dont want to copy and paste all the start nodes once for each possible input configuration that evaluates to the on state. Therefore, I used the <input> node in the <output> node to paste those start nodes inside each time.
However since the start nodes are a separate ROM now, the XML inside is no longer well-formed as it consists of multiple top-level <start> nodes. Wrapping them inside a <config> node breaks it, as the rom_filter also generates a <config> node and therefore does it double. Copying the start nodes inside no longer works.

Long story short, to support this we either need to select the sub node of the input ROM in the input node and be able to select this sub node in the <output><input/></output> construction, or we need some AND/OR operations inside the <if> statement.

What do you think of something like the following config syntax:
```
<if> 
  <or>
    <has_value input="foo" value="true"/>
    <and>
      <has_value input="bar" value="123"/>
      <has_value input="baz" value="456"/>
    </and>
  </or>
  <then> ... </then>
  <else/>
</if>
```
building this into the `Rom_filter::Main::_evaluate_node()` function with some recursive parsing of the <and> and <or> nodes should be doable.
Of course, in this case the original syntax of rom_filter would still be valid, making it backwards compatible.

BorisMulder-CSL, 2018-04-30:
this commit does that:
https://github.com/nlcsl/genode/commit/6849ab1375e895d6afc090fa4c08fb5ac4ea8ee1

BorisMulder-CSL, 2018-05-01:
These two commits added this feature.

nfeske, 2018-05-07:
Just as a quick heads-up, I won't be able to review this issue before the release of Genode 18.05. @BorisMulder-CSL Please bear with me. I hope that's ok for you.

nfeske, 2019-03-17:
The intermediate results entered the master branch in 18.05. This issue is stale since then. I'm not planning to come back to improving the ROM filter any time soon. Let's open new issues for further improvements.

closed 2019-03-17 by nfeske