genodelabs/genode #2693: fetchurl: truncate existing files

BorisMulder-CSL, 2018-02-26:
Fetchurl is given a path to a file it should download. Currently, when this file already exists, fetchurl will overwrite the existing file without truncating it. So if the new file is smaller than the old file, the last part of the old file will remain and the file becomes corrupted. An easy solution is to remove the old file before downloading it; I implemented this here: https://github.com/nlcsl/genode/commit/4477180e35c2871c8fb59ae1eccc89e96f2ee79e

Perhaps there is a better way to truncate the file at the end using ftruncate, however you need to know the length of the old and new file. What has your preference?

ehmry, 2018-03-01:
I would like to truncate to zero and then begin writing.

cnuke, 2018-03-01:
How ever the truncation is implemented, fetchurl should not override already existing files if not instructed to do so explicitly IMHO.

BorisMulder-CSL, 2018-03-01:
In such a case, there should be an option to turn this functionality on and off, truncating if on, and giving an error if off while trying to download an existing file.
This truncation becomes relevant e.g. if you want to do subsequent updates of the same file. There should not be a corner case where the file becomes corrupted in case the newest version is shorter than the oldest.

chelmuth, 2018-03-01:
Should be not that hard to implement in *fetchurl/component.cc* by using `O_EXCL` or `ftruncate` on file open. Happy hacking.

ehmry, 2018-03-01:
@BorisMulder-CSL FYI, a patch for a progress report and asynchronous fetches is in the works.

ehmry, 2018-03-02:
Unlinking an existing file may be an issue for some synthetic files. Unlink might be interpreted as a close request on a resource, but for the same file truncation would probably be a no-op. This is already the case that **run/fetchurl** is writing to an existing file at ``/dev/log``.

chelmuth, 2018-03-08:
FYI, issue #2702 was closed by today's master update but this does not include the file-truncation option as on @ehmry's topic branch.

nfeske, 2021-02-26:
Since no one seems to suffer under this problem in practice, I'm closing this issue.

closed 2021-02-26 by nfeske