genodelabs/genode #2695: Race condition when lseek()ing newly created file

senier, 2018-02-26:
When creating a new file using `open()`, writing data to it using `write()` and seeking to the end of the newly created file using `lseek(fd, 0, SEEK_END)`, there is a race condition. In the major number of cases the result is the length of data written, as expected. Sometimes `lseek()` returns 0.

When performing an `fsync()` after writing, the issue does not happen. This does not happen on Linux either.

[This](https://github.com/Componolit/componolit/tree/android_runtime/src/test/mfe/lseek_after_write) is a reproducer.

cnuke, 2018-02-27:
@senier please give c962d40 a try.

senier, 2018-02-27:
This fixes the issue (not unexpectedly, as a manual fsync() had the same effect). Thanks @cnuke for the quick resolution!

chelmuth, 2018-03-01:
I merged @cnuke's commit and also added another sync before ftruncate(). @senier do you approve?

senier, 2018-03-01:
Yes, I can confirm that the issue is fixed in staging. Thanks!

chelmuth, 2018-03-08:
Fixed on master branch.

closed 2018-03-08 by chelmuth