genodelabs/genode #2696: realpath() succeeds despite non-existing file

senier, 2018-02-27:
When calling realpath() on a file that does *not* exist, it should return NULL and set _errno_ to ENOENT (at least that's my [interpretation](http://pubs.opengroup.org/onlinepubs/9699919799/functions/realpath.html)). In Genode's libc, realpath() succeeds, even if the last component of the path does not exist.

Reproducer is [here](https://github.com/Componolit/componolit/tree/android_runtime/src/test/mfe/realpath_invalid).

chelmuth, 2018-02-28:
`realpath()` is a library function in *contrib/libc/src/lib/libc/lib/libc/stdlib/realpath.c*, so the issue could be in (from the `nm realpath.o` output):
```
                 U getcwd
                 U lstat
                 U readlink
```

nfeske, 2021-02-26:
I'm closing this issue because it has become stale.

closed 2021-02-26 by nfeske