genodelabs/genode #2697: [q] do I absolutely need gcc 6.3 and cannot use 6.4 ?

vladp, 2018-03-01:
Hello
newbie here, apologies if this is already covered somewhere.

I am getting
```
v@hp02_gnt ~/devel/other/genode/build/x86_64 $ make core server/nitpicker
/bin/bash: /usr/local/genode-gcc/bin/genode-x86-g++: No such file or directory
Makefile:131: *** "/usr/local/genode-gcc/bin/genode-x86-g++ version 6.3.0 is required".  Stop.
v@hp02_gnt ~/devel/other/genode/build/x86_64 $ 

```
My os is gentoo derivative and I just installed fresh copy that has a gcc 6.4
do I absolutely need to go back to 6.3 ?

thx

N8Fear, 2018-03-01:
Since the toolchain requires some modifications you should install the genode toolchain from https://genode.org/download/tool-chain (then you'll get genode-g++ which otherwise you won't have and which causes the actual error "No such file or directory"). The version will also be correct then.

senier, 2018-03-01:
Genode requires it's own tool chain which is independent of your system compiler. I'd suggest you have a look at section 2.3 of the [Genode Foundations book](http://genode.org/documentation/genode-foundations-17-05.pdf). The book, in general, is also a good way to get started with Genode.

Earlier in the same chapter you will also find a reference to the  mailing list, which is a better place to ask these kind of questions.

chelmuth, 2018-03-01:
@vladp if the comments above helped you solve your issue please close this tracker item.

vladp, 2018-03-02:
understood. thank you

closed 2018-03-02 by vladp