genodelabs/genode #2699: hw: avoid arm cache maintenance code duplication

skalk, 2018-03-01:
As @m-stein thankworthy recognized there is a huge code duplication in between `repos/base-hw/src/core/spec/arm_v7/cpu.cc` and `repos/base-hw/src/lib/hw/spec/arm/arm_v7_cpu.cc` that can be avoided.

skalk, 2019-03-13:
Related to this issue, Tomasz Gajewski kindly remarked on the mailing list that there are some artefacts in the usage of invalidate_*_cache currently use. Actually, we do not need the cache maintainance functions related to the whole cache in core/kernel anymore. A selective cache-flush is enough, now that the corresponding address-space is still the active one. Therefore, we can safely remove those functions, and only need to leave them in bootstrap to invalidate the whole cache in the very beginning.

m-stein, 2019-03-13:
Really cool. more than 200 LoC less in base-hw :)

commit 65fbdc7c (2019-03-19): hw: remove overall cache maintainance from core

This functionality is only needed in bootstrap now that kernel and
userland share the same address-space.

Fix #2699
commit e9b3569f (2019-04-01): hw: remove overall cache maintainance from core

This functionality is only needed in bootstrap now that kernel and
userland share the same address-space.

Fix #2699
closed 2019-04-02 by nfeske