genodelabs/genode #2700: Build problems with sculpt and demo

tomga, 2018-03-01:
Build was performed in a fresh environment where only sculpt for nova and foc was built earlier.

When trying to build sculpt on hw with:
```make -C build/x86_64 run/sculpt KERNEL=hw```

During first attempt I was asked to create depot for base-hw with:
```
/projects/genode/genode/tool/depot/create genodelabs/bin/x86_64/base-hw-pc CROSS_DEV_PREFIX=/usr/local/genode-gcc/bin/genode-x86-
```

Later I got following error:
```
make: Entering directory '/projects/genode/genode/build/x86_64'
including /projects/genode/genode/tool/run/power_on/qemu
including /projects/genode/genode/tool/run/log/qemu
including /projects/genode/genode/tool/run/image/iso
including /projects/genode/genode/tool/run/boot_dir/hw
including /projects/genode/genode/repos/gems/run/sculpt.run
checking configuration syntax
  CHECK init
core link address is 0xffffffc000000000
genode-x86-g++: error: debug/core-hw.o: No such file or directory
    while executing
"exec [cross_dev_prefix]g++ {*}$arch -nostdlib {*}[core_ld_opts]  -Wl,-z -Wl,max-page-size=0x1000  -Wl,-Ttext=$link_address -Wl,-gc-sections  -Wl,-nost..."
    (procedure "build_core" line 18)
    invoked from within
"build_core debug/$core_obj                 {} [run_dir].core      [core_link_address]"
    (procedure "run_boot_dir" line 43)
    invoked from within
"run_boot_dir $binaries"
    (procedure "build_boot_image" line 27)
    invoked from within
"build_boot_image $boot_modules"
    (file "/projects/genode/genode/repos/gems/run/sculpt.run" line 653)
    invoked from within
"source $include_name"
    ("foreach" body line 6)
    invoked from within
"foreach include_name [get_cmd_arg --include ""] {
	# first check if the include name is absolute
	if {[string first "/" $include_name] == 0} {
		puts ..."
    (file "/projects/genode/genode/tool/run/run" line 959)
Makefile:312: recipe for target 'run/sculpt' failed
make: *** [run/sculpt] Error 1
make: Leaving directory '/projects/genode/genode/build/x86_64'
```

Later I've tried to build demo which some time ago worked for me and I wanted to test it.
```
make -C build/x86_64 run/demo KERNEL=hw
```
Enabled and prepared some ports for which I was asked to make and build failed with:
```
genode build completed
using 'ld-hw.lib.so' as 'ld.lib.so'
using 'hw_timer_drv' as 'timer'
error copying "bin/fb_drv": no such file or directory
    while executing
"file copy -force bin/[kernel_specific_binary $binary] [run_dir]/genode/$binary"
    (procedure "copy_genode_binaries_to_run_dir" line 4)
    invoked from within
"copy_genode_binaries_to_run_dir $modules"
    (procedure "run_boot_dir" line 35)
    invoked from within
"run_boot_dir $binaries"
    (procedure "build_boot_image" line 27)
    invoked from within
"build_boot_image $boot_modules"
    (file "/projects/genode/genode/repos/os/run/demo.run" line 293)
    invoked from within
"source $include_name"
    ("foreach" body line 6)
    invoked from within
"foreach include_name [get_cmd_arg --include ""] {
	# first check if the include name is absolute
	if {[string first "/" $include_name] == 0} {
		puts ..."
    (file "/projects/genode/genode/tool/run/run" line 959)
Makefile:312: recipe for target 'run/demo' failed
make: *** [run/demo] Error 1
make: Leaving directory '/projects/genode/genode/build/x86_64'
```
but something that was built successfully in that step made the error from sculpt build go away and was able to build sculpt for hw. So probably either there is a missing dependency or something in sculpt build is used that should not be.

cnuke, 2018-03-01:
If you have followed the Sulpt EA guide, your `build.conf` has not all necessary repositories enabled to successfully execute `run/demo` on base-hw. You have to enable `libports` since `run/demo'  relies on the VESA frame buffer.

tomga, 2018-03-02:
Ok, I forgot about it. So the second problem is solved. I assume that this would not help for the first problem.

nfeske, 2018-03-02:
The first issue seems to be related to 7f5bec5c0dd5edec30c5d8dd463d3192974e9d0e, which apparently fixes the problem for most kernels (i.e., NOVA) but not for base-hw. @skalk What do you think?

@tomga I would not recommend using Sculpt on any kernel other than NOVA. Sculpt is used and tested only on this kernel. Other kernels may work for parts of it but not for the full scenario. In particular, without IOMMU, devices may violate address-space protection. Without hardware virtualization support, VirtualBox won't run. Genode supports these features on NOVA only. We have the plan to enable those features for base-hw and seL4 towards the end of the year.

tomga, 2018-03-02:
I've tried NOVA first. But all I get is:
{{{
Bender: Hello World.
framebuffer at [b8000+b8fa0) 80x25@16
warning - unknown framebuffer type
framebuffer at [b8000+b8fa0) 80x25@16
warning - unknown framebuffer type

NOVA Microhypervisor v7-3c86b31 (x86_64): Feb 28 2018 13:59:23 [gcc 6.3.0] [MBI2]
}}}
and reboot. Asking what to do about it was next on my todo but didn't want to mix issues.

Later I've tried hw but without any success due to build error. Next tried foc. This built and started but there where errors about "platform_info" (changed configuration for logs from drivers subsystem to get them on serial output)
{{{
[init -> drivers] Warning: dynamic: assigned RAM exceeds available RAM
[init -> drivers -> driver_manager] Error: ROM-session creation failed (ram_quota=6144, cap_quota=3, label="platform_info")
[init -> drivers -> driver_manager] Error: Could not open ROM session for "platform_info"
[init -> drivers -> driver_manager] Error: Uncaught exception of type 'Genode::Rom_connection::Rom_connection_failed'
[init -> drivers -> driver_manager] Warning: abort called - thread: ep
[init -> drivers] child "driver_manager" exited with exit value 1
[init -> drivers -> acpi_drv] Error: ROM-session creation failed (ram_quota=6144, cap_quota=3, label="platform_info")
[init -> drivers -> acpi_drv] Error: Could not open ROM session for "platform_info"
}}}
and it didn't succeded to show any graphical output.

Later I succeeded with build for hw and it started successfully sculpt with vesa framebuffer (I have nvidia on this machine).

How should I debug NOVA problems that cause reboot before starting any genode component?

alex-ab, 2018-03-02:
> How should I debug NOVA problems that cause reboot before starting any genode component?

I fear you will have to debug the NOVA kernel. I would suggest to use some simple Genode scenario like run/log instead of run/sculpt. Start instrumenting the NOVA kernel in ```contrib/nova-<hash>/src//kernel/nova/src/init.cpp``` after the last good known line 'NOVA Microhypervisor ...'

Whenever you rebuild the run/log scenario the kernel should also be re-build. We will need some clue what causes the issue from your side. Which hardware/CPU you are using ? Does a native Linux boot reveals something about suspicious ACPI tables ?

tomga, 2018-03-02:
I'll try to experiment during the weekend.

Some time ago I posted some patch to code in Genode for parsing ACPI tables on that machine (#1883) but that was in GGenode code not NOVA. Does NOVA do this independendtly during kernel initialization?

alex-ab, 2018-03-02:
> Does NOVA do this independendtly during kernel initialization?

Yes. Good luck.

skalk, 2018-03-02:
@nfeske yes you're right, I've overseen that base-hw special handling when preparing the fix, sorry therefore. I've prepared a follow-up commit e401a5c3478db78c7ea8df2a77d52c119f219910

tomga, 2018-03-02:
I've managed to fix reboot problem for run/log on NOVA with following patch:
```
--- contrib/nova-0878c653b740359a4123ac0b9a79ce5df26f9251/src/kernel/nova/src/dmar.cpp.orig	2018-03-02 20:56:07.215941095 +0100
+++ contrib/nova-0878c653b740359a4123ac0b9a79ce5df26f9251/src/kernel/nova/src/dmar.cpp	2018-03-02 20:58:48.501935354 +0100
@@ -82,11 +82,12 @@
     if (c->present())
         c->set (0, 0);
 
     flush_ctx();
 
-    c->set (lev | p->did << 8, p->dpt.root (p->quota, lev + 1) | 1);
+    if (c->present())
+        c->set (lev | p->did << 8, p->dpt.root (p->quota, lev + 1) | 1);
 
     p->assign_rid(rid);
 
     if (p != &Pd::kern && read<uint32>(REG_FECTL) & (1UL << 31)) {
         trace(TRACE_IOMMU, "DMAR:%p - re-enabling fault reporting", this);
```
I have no idea what this part of ACPI describes and if it is wrong on my machine but I think that patch is obviously ok due to the fact that few lines earlier access to ```c->set(...)``` is guarded by the same condition.

Possibly this part of ACPI data is related to printed later:
```
[ 0] disabling super pages for DMAR
```

Now I think I need to find out how to rebuild downloaded depot or somehow disable it to use my own NOVA kernel to be able to try sculpt on NOVA. Currently changes in source cause rebuild for run/log but not for run/sculpt.

alex-ab, 2018-03-05:
> but I think that patch is obviously ok due to the fact that few lines earlier access to c->set(...) is guarded by the same condition

Maybe it helps you, but the patch is wrong. With the line ```c->set (0,0)``` the entry becomes not present and after the flush_ctx, with your patch, you don't setup the correct DMAR (IOMMU) page tables for the device. It would be interesting which PCI device causes the trouble for you and find the actual reason.

If you don't want to patch the kernel, you may just disable the IOMMU/DMAR support by removing the ```iommu``` cmdline argmuent in tool/run//boot_dir/nova. However you will lose of course the protection provided by the hardware against misbehaving drivers and or devices.

tomga, 2018-03-05:
Thanks for the explanation. I'll try with disabling IOMMU as linux complains about ACPI tables during parsing DMAR on that machine too and disables IOMMU support although it is capable to go on after finding a problem.

alex-ab, 2018-03-08:
```
[init -> drivers -> driver_manager] Error: ROM-session creation failed (ram_quota=6144, cap_quota=3, label="platform_info")
[init -> drivers -> driver_manager] Error: Could not open ROM session for "platform_info"
```
e32b29c084864d5a4e16bf302815aeef8f77c368 helps for foc

tomga, 2018-07-06:
> If you don't want to patch the kernel, you may just disable the IOMMU/DMAR support by removing the iommu cmdline argmuent in tool/run//boot_dir/nova.

I've checked this and it helps. It allows to boot sculpt.

> It would be interesting which PCI device causes the trouble for you and find the actual reason.
I tried to search the web and it is possible that it is somehow related with integrated intel graphics - although it is not used by me as I have an nvidia card too.

Linux kernel shows in dmesg:
```
[    0.000000] ------------[ cut here ]------------
[    0.000000] WARNING: CPU: 0 PID: 0 at /build/linux-GU1w8g/linux-3.16.39/drivers/iommu/dmar.c:760 warn_invalid_dmar+0x7a/0x8a()
[    0.000000] Your BIOS is broken; DMAR reported at address fed90000 returns all ones!
BIOS vendor: American Megatrends Inc.; Ver: 1604   ; Product Version: System Version
[    0.000000] Modules linked in:
[    0.000000] CPU: 0 PID: 0 Comm: swapper Not tainted 3.16.0-4-amd64 #1 Debian 3.16.39-1
[    0.000000] Hardware name: System manufacturer System Product Name/P7H55D-M PRO, BIOS 1604    07/22/2010
[    0.000000]  ffffffff81017695 ffffffff81514c11 ffffffff81803e80 000000000000000b
[    0.000000]  ffffffff81068867 ffffffff81af401c ffffffff81803ed0 00000000fed90000
[    0.000000]  0000000000000000 ffffffff81803fb0 ffffffff81068914 ffffffff8176ed08
[    0.000000] Call Trace:
[    0.000000]  [<ffffffff81017695>] ? show_stack+0x5/0x50
[    0.000000]  [<ffffffff81514c11>] ? dump_stack+0x5d/0x78
[    0.000000]  [<ffffffff81068867>] ? warn_slowpath_common+0x77/0x90
[    0.000000]  [<ffffffff81068914>] ? warn_slowpath_fmt_taint+0x44/0x50
[    0.000000]  [<ffffffff8192f70a>] ? __early_ioremap+0x167/0x1fa
[    0.000000]  [<ffffffff8192f946>] ? early_ioremap+0xf/0x10
[    0.000000]  [<ffffffff81345364>] ? acpi_tb_acquire_table+0x39/0x66
[    0.000000]  [<ffffffff81515bf2>] ? warn_invalid_dmar+0x7a/0x8a
[    0.000000]  [<ffffffff81949c24>] ? detect_intel_iommu+0xe3/0x16a
[    0.000000]  [<ffffffff8190be1b>] ? pci_iommu_alloc+0x46/0x67
[    0.000000]  [<ffffffff8191b264>] ? mem_init+0x11/0x92
[    0.000000]  [<ffffffff81902e1a>] ? start_kernel+0x23b/0x4a2
[    0.000000]  [<ffffffff81902a04>] ? set_init_arg+0x4e/0x4e
[    0.000000]  [<ffffffff81902120>] ? early_idt_handler_array+0x120/0x120
[    0.000000]  [<ffffffff8190271f>] ? x86_64_start_kernel+0x14d/0x15c
[    0.000000] ---[ end trace 2358418464195214 ]---
```
and later
```
[    0.024055] dmar: Host address width 36
[    0.024056] dmar: DRHD base: 0x000000fed90000 flags: 0x1
[    0.024064] dmar: IOMMU: failed to map dmar0
[    0.024113] dmar: parse DMAR table failure.
[    0.024114] Switched APIC routing to physical flat.
```
@alex-ab if you think that it is worth to try to find a workaround for this to avoid reboot than I can try some your suggestions for fixes. If not then this issue can be closed.

alex-ab, 2018-07-09:
If you sent me by mail directly a dump of your ACPI tables, I may try to have a look. Maybe we can make the parsing a bit more robust (the hints from Linux trace are good!), but actually it as a BIOS/UEFI fault by providing bogus ACPI tables. (Any chance that there is a more recent BIOS update for you machine ?).

tomga, 2018-07-09:
Dump sent. No BIOS update available. At least I haven't found one.

alex-ab, 2018-07-11:
@tomga: thx. Can you please try the NOVA [commit](https://github.com/alex-ab/NOVA/commit/f8619ef50132c8aecf8c15c86e58e58eddfeef1b)

tomga, 2018-07-11:
I'll try but can you tell me what will be the easiest path for this? Do I correctly guess that changing `repos/base-nova/ports/nova.port` to this commit and later updating hash somehow is a good approach?

alex-ab, 2018-07-11:
If you want to do the Genode way, yes. Replace in ```repos/base-nova/ports/nova.port``` the commit with the one of the nova barnch. Afterwards invoke  ```tool/ports/update_hash nova``` which will update ```repos/base-nova/ports/nova.hash```
The you re-prepare you nova port by ```tool/ports/update_port nova``` and  re-compile&re-run.

The other option is that you just git cherry-pick the commit in the nova contrib sources and re-compile&re-run.

Your choice ;-)

tomga, 2018-07-11:
First check seems to be ok. I took "the other option", cherry-picked commit in nova contrib, performed: `make -C build/x86_64 run/demo KERNEL=nova`, dd and booted. Seemed ok - not rebooted.

I wanted to make sure everything is ok and tried `make -C build/x86_64 run/sculpt KERNEL=nova`, dd and ... probably I messed something up because after booting I have demo and not sculpt.

I don't understand how it's possible. I've checked that demo.img was built earlier, sculpt.img is bigger and that image extracted from pendrive gave exactly the same checksum of image. I don't feel comfortable with genode build system yet so maybe I've made something wrong. Anyway I'll try from scratch again in a few days and confirm it definitely fixes a problem.

tomga, 2018-07-15:
It seems that I had some problem with my linux writing to usb. Every time I attached it it showed in dmesg device /dev/sdc as attached. After dd led blinked on a pendrive. Reading from that pendrive seemed to read just what I previously wrote but it seems the write was not performed somehow.

Reboot of linux and writing image again helped (normally I just suspend this system).

I was able to boot sculpt after applying fix with "the genode way" [(commit)](https://github.com/tomga/genode/commit/b9f5303ffa6ef9245133817cb0cf95cfaf92b6a6) so the fix is working.

Thanks.

alex-ab, 2018-07-17:
@tomga: Good. I updated the current kernel branch (the one here used for 18.05 is already outdated). The commit should show up the next time we master Genode (or next release). Thanks again for your effort.

@cnuke: please add  4d98930f4ae91e3767b4d765cf902cefa5fbe24d to staging

commit c2e1ee55 (2018-08-02): nova: update kernel branch (MCA, invalid DMAR)

to avoid resetting due to invalid IOMMU/DMAR hardware units.

Fixes #2700
closed 2018-08-08 by nfeske