genodelabs/genode #2701: Pass root directory to VFS plugings

nfeske, 2018-03-02:
In a growing number of VFS use cases, it is desirable to allow VFS plugins to use the VFS. This would ultimately allow the chaining of plugins.

ehmry, 2018-03-27:
Perhaps the number of `Vfs::File_system` constructor arguments could be reduced with something like a `Vfs::Env` object with a `root_dir` accessor?

```
struct Vfs::Env
{
	virtual Genode::Env &env() = 0;
	virtual Genode::Allocator &alloc() = 0;
	virtual Vfs::Io_response_handler &io_handler() = 0;
	virtual File_system &root_dir() = 0;
	virtual Timeout_scheduler &timeout_scheduler() = 0;
};
```

What I really want is to sneak in the `timeout_scheduler`, which would increase the efficiency of timeouts in the VFS by using a single scheduler rather than creating multiple timer sessions. In the case of the libc VFS and a non-libc VFS, the timer connection would be lazily constructed.

nfeske, 2018-03-28:
@ehmry introducing the `Vfs::Env` is a good idea. Let us do this as a subsequent step.

The remark about the timeout scheduler is interesting. I haven't had this problem so far but as more and more features will be implemented as VFS plugins in the future, I can see how this makes sense. That said, I find it important to not make the VFS inherently dependent on a timer session. So the `timeout_scheduler` must be constructed lazily - as done for timer session in the libc.

nfeske, 2018-03-30:
Merged to staging.

commit f4403e5b (2018-03-30): vfs: pass root directory to plugins

This patch enables the use of the VFS from VFS plugins by passing a
reference of the root directory to the constructors of file-system
instances. Since it changes the signature of 'Vfs::Dir_file_system',
any code that uses the VFS directly requires an adaptation.

Fixes #2701
commit b0b92e4e (2018-04-10): vfs: pass root directory to plugins

This patch enables the use of the VFS from VFS plugins by passing a
reference of the root directory to the constructors of file-system
instances. Since it changes the signature of 'Vfs::Dir_file_system',
any code that uses the VFS directly requires an adaptation.

Fixes #2701
closed 2018-04-10 by chelmuth