genodelabs/genode #2702: fetchurl progress report

ehmry, 2018-03-02:
The #2684 patch removed progress logging from fetchurl and now the utility is too opaque about what is happening. The log session should be for logging events, not status, so a status report would be more appropriate.

ehmry, 2018-03-02:
Fixed on my **issue2702_fetchurl_progress** branch following @BorisMulder-CSL's patches.

ehmry, 2018-03-02:
Fixed to use standard `<report progress="..."/>` pattern at  516777f881cac5c09b00e20f075863765a95b974.

chelmuth, 2018-03-05:
Thanks, merged your commit (plus one from @BorisMulder-CSL). I skipped the assert/remove commit, though, as I still fancy the truncate solution more.

ehmry, 2018-03-05:
@chelmuth I have a small fixup for printing failed URLs with error messages at b03fcd7b839753eeabfa2f6dc00488bba4f0c0ae.

closed 2018-03-08 by chelmuth

ehmry, 2018-03-21:
I have a minor commit 077818d284d8c8f03cc1f05bdc56c255428715bd to fix the fetchurl package.

commit 07e2487c (2018-03-05): Fetchurl progress reporting

Refactor the fetchurl utility to optionally report the initial fetch
state, fetch progress, and the final state.

Fix #2702
commit 84ac5891 (2018-03-08): Fetchurl progress reporting

Refactor the fetchurl utility to optionally report the initial fetch
state, fetch progress, and the final state.

Fix #2702