genodelabs/genode #2704: Handling timeouts as a side-effect of scheduling one-shot-timeouts

cnuke, 2018-03-04:
Whenever a new one-shot-timeout is scheduled, https://github.com/genodelabs/genode/blob/1eefc8d42d2638a5bc483c105356c3b4aa41f801/repos/os/src/lib/timeout/timeout.cc#L105
the handler of an already scheduled timeout may be called https://github.com/genodelabs/genode/blob/1eefc8d42d2638a5bc483c105356c3b4aa41f801/repos/os/src/lib/timeout/timeout.cc#L112

which will lead to a deadlock if the executed handler access' the same locks as the code scheduling the new timeout.

m-stein, 2018-03-21:
Fixed:
7b55d11 test/timeout: test locks in handlers
69e29e3 test/timeout: static constructors and typo
983ecb9 timeout: do not handle timeouts while scheduling
e82723d timeout: dequeue all pending alarms before reschedule
faee7e9 timeout: _setup_alarm calculates deadline itself
16c8c4f timeout: become independent of the Alarm framework

Missing:
* Simplify and optimize Timeout framework even further now without the intermediate Alarm layer
* Remove Alarm framework in general

m-stein, 2018-03-30:
Updated according to suggestions by @chelmuth:
eb52c81 test/timeout: test locks in handlers
5949df0 test/timeout: static constructors and typo
8cf7153 timeout: do not handle timeouts while scheduling
531aaef timeout: dequeue all pending alarms before reschedule
8921713 timeout: _setup_alarm calculates deadline itself
79189a1 timeout: become independent of the Alarm framework

m-stein, 2018-04-05:
I'm at it.

m-stein, 2018-04-05:
Uploaded a fix:
4ece6d6 timeout: consider age of local time value

To my shame, this regression would have been forseeable. I knew that the local time would always be a little bit out of date when removing the hack of updating time during schedule. But I didn't imagine that it would have such significant effects.

m-stein, 2018-04-05:
I successfully tested terminal_echo, part_blk, part_blk_gpt, timer and timeout with the fix.

commit bd4823fe (2018-04-03): test/timeout: test locks in handlers

Originally, the timeout framework caused deadlocks when acquiring the same
lock from different timeout handlers. This use case is now tested in the
timeout test.

Fixes #2704
closed 2018-04-10 by chelmuth