genodelabs/genode #2705: bash port doesn't support common navigation keys

NobodyIII, 2018-03-05:
Navigation keys such as Home, End, and Delete don't work in bash, but instead print the tilde ('~') character. How can we fix this?

chelmuth, 2018-03-08:
A starting point for supporting more terminal capabilties in bash's readline implementation is to remove `-DNO_TTY_DRIVER` from *repos/ports/src/noux-pkg/bash/target.inc* and fix all subsequent issues.

nfeske, 2018-03-16:
@NobodyIII As far as suitable topics for contributions go, this issue would be a perfect candidate. The problem needs definitely to be fixed because it leaves a bad impression to new users. But since is not on the critical path for the regular developers, it remains unfixed. Would you like to take on it? A fix would be very much appreciated!

NobodyIII, 2018-03-16:
I'll get started on it now.

NobodyIII, 2018-03-16:
It looks like I just need to insert a working /etc/inputrc into bash.tar.
How do I do that? How do I add a file to a noux package?

nfeske, 2018-03-16:
To test this idea, you may simply add an '<inline>' node to the VFS used by noux.

NobodyIII, 2018-03-16:
> To test this idea, you may simply add an '' node to the VFS used by noux.

I've already tested that it works. I just need to include the file.

chelmuth, 2018-03-16:
Did you actually fix the issue of the missing TTY driver or did you just add some control sequences to *.inputrc*? IMO, the latter is not desirable and no valid fix.

chelmuth, 2018-03-16:
On second thought and after consulting some documentation/howtos for readline, I get the feeling the implementation may lack automatic adaption of key (escape sequences) according to the running terminal... So, we need more investigation first.

NobodyIII, 2018-03-17:
Adding my Linux system's /etc/inputrc (from Debian's readline-common) to
bash.tar worked perfectly. It works, and it's the standard solution, as far
as I can tell.

As for the TTY driver, you can remove the -DNO_TTY_DRIVER, and bash
compiles just fine.

commit 170507a8 (2018-07-03): bash: support home and end keys

This commit contains a patch of bash's built-in default escape sequences
to match the 'screen' terminal as used by Genode's terminal component.

Fixes #2705
closed 2018-08-08 by nfeske