genodelabs/genode #2707: vfs lxip: File_system session resource leak

BorisMulder-CSL, 2018-03-05:
In my scenario, I run fetchurl on top of init, similar to the sculpt scenario, except that I periodically poll for updates from a remote server.
So far it works fine, but when I let it run for a while, sometimes the socket vfs does not release the resources from the lxip File_system session. This only randomly happens after it has started and stopped fetchurl 20+ times.
The log gives the following error:

```
Warning: PD (init -> vfs) RAM limit (used=30276K, limit=31085713) exceeded during transfer_quota(194253)
[init] Warning: init_system: could not revert session RAM quota (service=File_system cid=<unconstructed> args='tx_buf_size=131072, root="", writeable=1, cap_quota=12, label="init_system -> init_fetchurl -> fetchurl -> socket", ram_qu
[init] ota=194253, diag=0' state=CLOSED ram_quota=194253, cap_quota=12)
Warning: PD (init -> init_system) cap limit (used=240, limit=248) exceeded during transfer_quota(12)
[init_system] Warning: init_fetchurl: could not revert session cap quota (service=File_system cid=735 args='tx_buf_size=131072, root="", writeable=1, cap_quota=12, label="init_fetchurl -> fetchurl -> socket", ram_quota=195038,
[init_system] diag=0' state=CLOSED ram_quota=195038, cap_quota=12)
```

It keeps leaking caps and RAM and after a while fetchurl will not restart as it is out of ram or caps, whichever has the most quota. If I look at the init report, you can see it loses 12 caps and 194-195k RAM each time, exactly the numbers it uses for the file_system session.

This was the issue on 17.11, and updating to 18.02 did not fix it.

ehmry, 2018-03-05:
I suspect that sockets might be left open when the client disconnects from the VFS server, maybe sockets should be closed automatically when no handles are held on the control files?

ehmry, 2018-03-15:
I am drafting a patch to make socket cleanup more explicit by requiring the socket_fs user to hold onto the `new_socket` file for the lifetime of the socket. That means a new socket is created each time the  `new_socket` is opened, and the same socket path will always be returned when reading the same handle on `new_socket`. This implies that when a session is closed at the VFS server, any remaining open sockets created during the session will be freed when the server closes all remaining open handles.

This much is implemented, however the accept procedure is not clear to me. Perhaps reading the `accept` file will return the remote address of the next pending connection will be returned, and a `accept_socket` control file would be added to the socket directory that would behave as described for the protocol level `new_socket`.

chelmuth, 2018-03-15:
Note that if the life time of a socket is bound to the life-time of the open *new_socket* handle, socket fs can no longer be used from noux/bash. Maybe we could drop this use case but I can remember a vivid discussion about it in the past. @cnuke how does Plan9 manage socket life time?

ehmry, 2018-03-15:
This would complicate `dup`/`dup2` as well. Maybe a reference counting scheme would work instead.

chelmuth, 2018-03-15:
IMO `dup()/dup2()` should be implemented locally in libc as the file-system session has no handle state that must be duplicated. Actually it seems to me that `dup()` and `dup2()` might be broken currently, though each function in a slightly different way. 1) VFS plugin does not implement `Plugin::dup()` and I couldn't find a generic implementation and 2) VFS plugin misses reference counting for duplicated open VFS handles, whcih means the first `close()` invalidates both libc file descriptors. If I'm mistaken please correct me as I had a quick look only.

ehmry, 2018-03-20:
I've implemented the lifetime mechanism described earlier in patch 058243acf1cb34a7f13490d610e220187f6be60e. No changes were made to Noux, but I think some special handling in Noux to keep sockets alive may be acceptable.

chelmuth, 2018-04-10:
@ehmry do you confirm 058243acf1cb34a7f13490d610e220187f6be60e is fit for staging? Did you look into `dup()` regarding my reservations about the current implementation too?

ehmry, 2018-04-12:
Yes, I believe the patch is ready for staging, rebased at d4c0ef48b742717f8c9679f210d79d4c7192445a. I have my doubts that `dup` behaves as it should, but without porting something like inetd, it shouldn't be an issue with sockets. Note that this patch will complicate making Noux work with socket_fs.

ehmry, 2018-09-11:
I believe this issue can be closed now that socket lifetimes are automatically managed.

closed 2018-09-11 by chelmuth