genodelabs/genode #2711: Link EGL to the Mesa library

ehmry, 2018-03-10:
The EGL library allows application to avoid explicit knowledge or requirements of Mesa with the `eglGetProcAddress` procedure.

ehmry, 2018-03-10:
Components that use GL headers still need to list `mesa` in LIBS or use `INC_DIR += $(call select_from_ports,mesa)/include/mesa`. Perhaps this can be solved with a header-only GL library?

chelmuth, 2018-03-14:
@ssumpf @cnuke opinions?

ssumpf, 2018-03-14:
@ehmry : It should suffice to add `mesa_api` to `LIBS` in order to get access to the OpenGL headers.

ehmry, 2018-03-14:
In that case the commit message should probably be changed to say something like
> Components may wish to list 'mesa_api' in LIBS to access OpenGL headers.

chelmuth, 2018-03-14:
Adapted the commit message and force-pushed staging 98ad7961eefcbb51112fc4b6cc415260b8cf4c26.

ehmry, 2018-03-14:
Cool, staging now supports the libretro frontend passing a GL context to cores.

commit e68fdda4 (2018-03-14): Link EGL to Mesa

Link the EGL library with Mesa. SDL's `GL_GetProcAddress` now calls
`eglGetProcAddress` rather than lookup procedures directly in the Mesa
library.

Components may still wish to list 'mesa' in LIBS to access OpenGL
headers.

Fix #2711
commit 98ad7961 (2018-03-14): Link EGL to Mesa

Link the EGL library with Mesa. SDL's `GL_GetProcAddress` now calls
`eglGetProcAddress` rather than lookup procedures directly in the Mesa
library.

Components may still wish to list 'mesa_api' in LIBS to access OpenGL
headers.

Fix #2711
commit 4b165190 (2018-03-27): Link EGL to Mesa

Link the EGL library with Mesa. SDL's `GL_GetProcAddress` now calls
`eglGetProcAddress` rather than lookup procedures directly in the Mesa
library.

Components may still wish to list 'mesa_api' in LIBS to access OpenGL
headers.

Fix #2711
closed 2018-03-29 by chelmuth