genodelabs/genode #2713: regression: ps2_drv no longer works with Dell Latitude 6430u touchpad

NobodyIII, 2018-03-13:
When I ran Sculpt on my laptop, I found that it didn't accept any input from the touchpad, unlike Turmvilla, which had handled it just fine. I eventually found the source of the regression: commit 7729398 - ps2_drv: reset mouse at startup

How can we fix this without reintroducing the problem that was solved by that commit? Simply using CMD_RESET then CMD_SET_DEFAULTS doesn't work. When reading the Sculpt log with the 18.02 ps2_drv driver, I find "could not reset mouse (unexpected response)".

nfeske, 2018-03-13:
It seems that each vendor has a differently broken PS/2 emulation. The mentioned commit was actually the result of hours of experimentation by @cproc. The SET_DEFAULTS command reliably broke the trackpoint support. With the RESET command in place, it magically worked. Still, the emulated PS/2 mouse fails to respond correctly (at least from our understanding of the PS/2 protocol) to the reset command. This is why we are seeing the warning message. However, the emulated mouse works after the reset. That said, even with the patch, the trackpoint support is still flaky. In particular, when configuring the BIOS to save power when on battery, the trackpoint never works.

I am currently out of ideas how to solve this problem in a systematic way. As last resort, we could inspect the initialization sequence of a driver that is known to work everywhere. Maybe we are lucky with Linux?

NobodyIII, 2018-03-13:
The Linux driver is probably the best place to look for a working
initialization sequence.

N8Fear, 2018-03-13:
From a quick look (Linux 4.15.9) they seem to send 'PSMOUSE_CMD_RESET_BAT' which is '0x02ff' to get a reset from a generic PS/2 mouse (or synaptic touchpad). In the Genode Code 'CMD_RESET' is just '0xff'.
The initialization of a trackpoint is different: here they send 'TP_COMMAND' ('0xE2') followed by 'TP_POR' ('0x7F') - and that up to two times (if the result is '0xfc' and '0x00').
After that they check for '0xaa' and '0x00', just like in the Genode code.

chelmuth, 2018-03-19:
@NobodyIII did you try to adapt the PS/2 driver according to @N8Fear's findings? Does your touchpad work again?

NobodyIII, 2018-03-19:
I tried it, but it didn't work.

chelmuth, 2018-08-17:
@NobodyIII did cleanup and preserve your fix? I discovered the regression with a different notebook's touchpad too.

chelmuth, 2018-08-17:
Commit dd60fbae4a3e1249a52fcde9ae63289119f87b7a implements a first attempt to tackle the issue but is tested on the *different* notebook mentioned above only. I appreciate any test results on other hardware, esp. regarding the needed timeout printed in the log:

```
[init -> ps2_drv] Using keyboard with scan code set 1 (xlate)
[init -> ps2_drv] >>>DEBUG<<< timeout_ms=400
[init -> platform_drv] PS2 uses IRQ, vector 0x1
[init -> platform_drv] PS2 uses IRQ, vector 0xc
```

chelmuth, 2018-08-20:
I tested the commit on a range of hardware (x250, Fujitsu notebooks, custom PCs) with no issues and the timeout never exceeded 450 ms.

chelmuth, 2018-08-20:
Seems x260 (with and without AC) can be added to the list of working devices.

NobodyIII, 2018-08-21:
I have confirmed that the commit fixes the problem on my Dell Latitude 6430u laptop. Thanks!

commit da3db5f6 (2018-08-22): input/ps2: poll for mouse-reset results

The PS/2 driver retries to get mouse-reset results for 700 ms, sleeping
after each attempt for 10 ms. So, the driver needs a Timer session now.

Fixes #2713