genodelabs/genode #2714: libc:

nfeske, 2018-03-14:


nfeske, 2018-03-14:
Please note the updated issue description.

ehmry, 2018-03-14:
This seems like a good time to raise the issue of **st_blksize**, if the VFS were to support this information it could inform the behavior of the internal read loop as well as the application. How much it would effect performance, I'm not sure.

> The size-related fields of the struct stat are as follows:
>   st_size	 The file size in bytes.
>   st_blksize	 The optimal I/O block size for	the file.
>   st_blocks	 ...

https://www.freebsd.org/cgi/man.cgi?query=stat&sektion=2&manpath=FreeBSD+11.1-RELEASE+and+Ports

cnuke, 2018-03-14:
It stands to reason if we need to fix it at all and how to do so. The underlying file system session does not provide any means to gather this information currently and would need to be extended. Besides ported POSIX programms. e.g. `cat(1)` adjusting its buffer size, I see no immediate use-case to justify the change and programms that deal with block devices will probably revert to using the ioctl interface anyway. From what I can tell, other systems choose a somewhat arbitrary value  (Linux uses `4K` while OpenBSD sets it to `16K`) that is a multiple of the block size. That being said, increasing the default I/O block size to `4096` as well on our part should not do any harm.

ehmry, 2018-03-27:
This is maybe something for another issue, but the Fs_file_system write implementation in the VFS does not gracefully handle the case where the size of a write buffer exceeds the size of the packet buffer. The Fs_file_system is the case where *blksize* information would be most useful.

ehmry, 2018-03-31:
I like the idea of bigger read ops, when dumping large data into terminal sessions I've noticed that each operation can be quite small, less than 4KiB.

nfeske, 2019-11-06:
Superseded by #3507

closed 2019-11-06 by nfeske