genodelabs/genode #2715: seoul: update to work with recent Linux VMs and to be usable by Sculpt scenario

alex-ab, 2018-03-14:


alex-ab, 2018-03-15:
All commits on [issue branch 2715](https://github.com/alex-ab/genode/commits/issue_2715) without the WIP tag are fine for review, @chelmuth, @nfeske - please consider for staging

alex-ab, 2018-03-21:
@chelmuth: please consider https://github.com/alex-ab/genode/commit/1e39c4429370db91a677f13400e158eb863aa242, https://github.com/alex-ab/genode/commit/08675019ffc875697edb739691b6e0392f2fbe1a and https://github.com/alex-ab/genode/commit/13d1fd5e9d9fa41885c152be7877ea18ea97c604 for staging

alex-ab, 2018-04-13:
I published a working version of Seoul for Sculpt with a recent Tinycore VM just containing a mostly recent Firefox browser. 306970891f7396584f42a4f0ebdeae37fb905816 and c2cb9e3eb4421e48f7e6f750e49f185b5a0b7de4 and e8becd69dd222783b8f76f5e93a0d1867e357b96 showcases the things which one has to add to a already running Sculpt scenario.

alex-ab, 2018-05-09:
Same as for Seoul I now added a version for Virtualbox5 containing just a small (disposable non-persistent) Tinycore/Firefox VM package.  Updated all in Tinycore and Firefox to version 60 and published it - unfortunately I was not able to reuse the same image at moment.

nfeske, 2018-08-22:
Fixed in master.

closed 2018-08-22 by nfeske