genodelabs/genode #2725: Gems: magic ring buffer

ehmry, 2018-03-18:
I have a ring buffer that I use in some of my side projects that I would like to have in the Gems repo.

ehmry, 2018-03-18:
Patch at 3edeaa96fd99e2e00c73d24d3bc39d129f196fcc.

nfeske, 2018-03-19:
That's a cool idea. It would fit well in the gems repository. I have the following remarks about the commit.

* It would be good to apply the common indentation style where the `public:` and `private:` section labels are indented.
* Each of both `attach_at` may fail with an exception (e.g., when multiple threads interact with the address space concurrently). To make the code more robust, it should be wrapped in a try-catch block. The handling code should revert the (potentially successful) first attach before re-throwing the exception.
* In the error case, the constructor just throws an `Exception`. How about throwing a more descriptive exception type like `Region_map::Region_conflict`? It would be good to document the exception in a comment of the constructor (using the `\throw` notation).
* I would prefer to avoid the term `PAGES` at the API level. Couldn't the user of this utility just specify a number of bytes as template argument instead?

chelmuth, 2018-03-19:
It seems `PAGES` is just an artifact and should read `SIZE` as in the implementation.

It may sound like overkill but the ring-buffer region could be reserved by using a managed dataspace and attaching the RAM dataspaces in the nested `Region_map`.

ehmry, 2018-03-19:
Yes, PAGES was an artifact. This utility is a bit dangerous because if the size TYPE is not a power of two the buffer becomes trashed after SIZE bytes have been written.

Fixup at e31cd054cc9efe90d3cf327d5c1b0dca5a32c0ed.

nfeske, 2018-03-19:
@ehmry Thanks for the quick overhaul.

nfeske, 2018-03-30:
Merged to staging.

ehmry, 2018-04-04:
@chelmuth do you have an example of reserving address space with managed dataspaces?

chelmuth, 2018-04-04:
@ehmry The *sub_rm* test seems to be a good starting point. Please refer to https://github.com/genodelabs/genode/blob/master/repos/base/src/test/sub_rm/main.cc#L64

ssumpf, 2018-04-04:
@ehmry:
`Rm_connection _rm_connection { env };`

/* create region map (reserve address space */
`Region_map_client _rm { _rm_connection.create(<size>);`

/* attach map to global region map */
`env.rm().attach(_rm.dataspace());`

/* attach something into region map */
`_rm.attach(ds);`

ehmry, 2018-04-04:
I have a patch to improve the buffer at 86e1abd1f5db52c9c22060965ffa5294127b15d2. The buffer now uses a region map to reserve space and comes with a test.

commit e60e0820 (2018-03-30): Gems: magic ring buffer

A ring buffer that uses a single dataspace mapped twice in consecutive
regions. This allows any operation that is less or equal to the size of
the buffer to be read or written in a single pass.

Fix #2725
closed 2018-04-10 by chelmuth