genodelabs/genode #2726: Support system reset via x86 ps2_drv

chelmuth, 2018-03-19:
In issue #2547 (https://github.com/genodelabs/genode/pull/2547#issuecomment-364647403) @nfeske proposed to implement a simple reset mechanism via the *magic* keyboard-controller bit in *ps2_drv*.
> extend the PS/2 driver with the option to listen to the "system" ROM module and trigger a reset at the ps2 controller once the state changes to "reset".

nfeske, 2021-03-29:
Fixed by the series of commits shown above.

alex-ab, 2021-04-15:
I discovered commit b9a5053ae1a51cd291503c3793d4f09e6e24e17c, which re-introduce the bug of #3829. The specific table may be shorter than the members used to access the table. Additionally, the commit misses to check the type of the reset, which is not ever of type IO port. If solely the I/O type should be supported, then the register values should not be reported to consumers of the report.

alex-ab, 2021-04-15:
> Thanks for your review. Do you happen to have a way to reproduce the out-of-bounds issue?

No, according to the mailng list discussion of #3829 you need Windriver Simics.

> The commit is still on staging. I wonder, do you already have some kind fixup in the pipe?

No, but I can offer to come up with fixups.

nfeske, 2021-04-15:
> No, but I can offer to come up with fixups.

That would be really nice. Thank you for the offer.

alex-ab, 2021-04-15:
I added to fixups, please review.

nfeske, 2021-04-15:
Thanks @alex-ab for the thorough rework. I cherry-picked both commits to staging and will give the mechanism a try on my machine tomorrow.

nfeske, 2021-04-16:
Just want to report back that the ACPI reset still works fine with the latest commit on my x250. :-)

alex-ab, 2021-04-16:
Sounds good :-)

nfeske, 2021-05-28:
Fixed in master.

closed 2021-05-28 by nfeske