genodelabs/genode #2728: nit_fb: add optional transparency

NobodyIII, 2018-03-20:
Nitpicker supports transparency, and I propose that we extend that to nit_fb. I admit that there aren't very many obvious use cases for this, but it will help me debug my image renderer's transparency support.

I'm working on this myself, and I'll probably have a patch soon.

NobodyIII, 2018-03-20:
I have this working with commit bedadbb in my nit_fb_transparency branch.

nfeske, 2018-03-21:
That's a good suggestion. Thanks for posting the commit. This is certainly a good starting point. For the real integration, we should also consider that the alpha attribute might be toggled dynamically. Whenever this happens, the client should receive a mode-changed signal.

I'd prefer to call the attribute just 'alpha' because it is shorter. Also internally, we follow the convention to not prefix boolean members with `is_` or `has_` unless it sounds awkward otherwise. In this case, I'd call the member just `_alpha`.

ehmry, 2018-03-23:
I'm drawn to this issue because I would like to request mode changes at the client-side and this could fit with requests for alpha. For my use case it would be sufficient to make synchronous and stateless requests and handle a mode change from my existing signal handlers. If the new mode comes with an alpha buffer, it seems to me that this would best be conveyed in the `Framebuffer::Mode` object when the client calls `Framebuffer::mode()`.

nfeske, 2018-03-23:
@ehmry Would it be possible to request a mode change by re-constructing the `Framebuffer::Connection`? I would prefer this over extending the framebuffer session interface.

nfeske, 2018-03-23:
BTW, I agree that adding the alpha attribute to `Framebuffer::Mode` makes much sense.

ehmry, 2018-03-23:
@nfeske yes, that is probably sufficient, my only concern is speed, but I will have to test it before I make any complaints.

chelmuth, 2019-03-14:
@NobodyIII are you still working on this feature? Did you already address @nfeske's remarks in https://github.com/genodelabs/genode/issues/2728#issuecomment-374874467?

nfeske, 2019-03-14:
As nit_fb is going to become irrelevant in the light of https://github.com/genodelabs/genode/issues/3196. So I would not spend any energy on it.

NobodyIII, 2019-03-15:
I don't remember the state of my code. I have too many other things that
I've been focusing on.

nfeske, 2019-03-15:
@NobodyIII do you agree to close this issue?

NobodyIII, 2019-03-15:
Yes.

closed 2019-03-15 by nfeske