genodelabs/genode #2729: false comment in reinterpret_cap_cast (from base/capability.h)

NobodyIII, 2018-03-22:
While reading through base/capability.h to better understand Genode, I came across the following false comment:

	/*
	 * The object layout of untyped and typed capabilities is identical.
	 * Hence we can just use it's copy-constructors.
	 */

This property isn't relied on at all in the actual code.

For reference, here's the function:

    /**
     * Convert an untyped capability to a typed capability
     */
    template <typename RPC_INTERFACE>
    Genode::Capability<RPC_INTERFACE>
    Genode::reinterpret_cap_cast(Untyped_capability const &untyped_cap)
    {
	    /*
	     * The object layout of untyped and typed capabilities is identical.
	     * Hence we can just use it's copy-constructors.
	     */
	    Untyped_capability *ptr = const_cast<Untyped_capability*>(&untyped_cap);
	    return *static_cast<Capability<RPC_INTERFACE>*>(ptr);
    }

Also, should this really include a const_cast? Either this should be removed, or the description should say something like "Convert a const untyped capability to a mutable typed capability".

skalk, 2019-03-12:
@NobodyIII can you please amplify why do you think that untyped and typed capabilities do not share the same object layout? I cannot see any additional members in Capability nor that it extends other classes than Untyped_capability.
To be honest the signature clearly states that you put a const capability into the function and gain a capability by value. I would find a annoying as reader of an API to always get confronted with the obvious. Because the source value you hand over is a reference, it clearly should be marked as const. The return-value is given by (a copied) value, so it is clearly mutuable. If we always put that adjectives in all function descriptions, there won't be much exceptional cases.

skalk, 2019-03-14:
I close this issue for now. If there is still a need for discussion, we can still do that here.

closed 2019-03-14 by skalk