genodelabs/genode #2732: NIC router: ICMP "Echo" and "Destination Unreachable"

m-stein, 2018-03-23:


m-stein, 2018-03-29:
Fixed:
7e773b7 autopilot.list: add ping_nic_router test
32118b0 ping_nic_router.run: ICMP through a nic_router
3760e08 nic_router: forward ICMP dst unreachable messages
9704633 nic_router: print packet info before sending it
f59a06e nic_router: config attribute 'mac_first'
fa27d8d nic_router: NAPT for ICMP echo messages
fe166e0 nic_router: fix bug in Link::handle_config
47b98c2 nic_router: conform log in Interface::_handle_eth
693a598 nic_router: packet verbose at detached interfaces
ef07326 nic_router: send ICMP error on unroutable packet
33f4e0f nic_router: fix missing IPv4.ECN initialization
1249217 autopilot.list: add ping test
345b117 app/ping: perform ICMP Echo to another IP host
19ad353 lwip: enable ICMP support
2e05823 nic_dump: support ICMP
04f95d6 net: ICMP protocol

chelmuth, 2018-04-03:
Currently, ICMP is not documented in the README or am I missing something?

m-stein, 2018-04-03:
@chelmuth You're right. I'll fix that.

commit c925cfd8 (2018-04-04): autopilot.list: add ping_nic_router test

Fixes #2732
closed 2018-04-10 by chelmuth