genodelabs/genode #2735: trace buffer does not wrap correctly

ValiValpas, 2018-03-28:
When the trace buffer wraps, the last entry cannot be detected anymore. The last committed entry should always be followed by an entry with len=0.

nfeske, 2018-04-03:
Thank you for fixing this issue and for extending the test. @m-stein Could you please have a look?

m-stein, 2018-04-03:
@valivalpas Thanks from my side too! Apart from the inline comments that I wrote, the commit seems good to me. However, I don't have a good feeling with overloading the meaning of _Entry::len and _Entry::last(). There may now be two last entries in a buffer with different meanings which must be kept in mind by users and developers. But I have the feeling that this should be part of a general review of the trace buffer structure in the future which was already suggested by https://github.com/genodelabs/genode/issues/2654#issuecomment-361745416.

ValiValpas, 2018-04-09:
@m-stein Thanks for the comments. I agree with splitting up the commit; are you going to do this?

Regarding your bad feeling: I was wondering about the semantics of `Entry::last()` anyway as it is implemented more like an `Entry::invalid()`. The `next()` function returns an invalid Entry object once the object is already invalid, the length is 0, or we hit the end of the buffer. Do you suggest to use the len field as the ultimate end-of-buffer indicator instead? 

In the current implementation, the `next()` function does not wrap around by itself, i.e. the TRACE client must take care of this by reading the two parts separately, once the buffer wrapped. So, what about making the `next()` function return all the entries until the logical end buffer (not the end of the buffer memory).

m-stein, 2018-04-10:
Reworked the commits:
1f7dd49 autopilot.list: add trace_logger test
e388293 test/trace_logger: fully automate test
b9a1acc trace_logger: be aware of trace-buffer wraps
01c1878 trace/buffer: keep "last" entry on wraps

@ValiValpas I fixed all that I commented plus the finish string pattern and the loop timeout in the test. I also documented the double-meaning of Trace::Buffer::Entry::last. I set the author of the new commits to you and added me as co-author, I hope this is the intended way?

ValiValpas, 2018-04-10:
@m-stein Perfect!

chelmuth, 2019-03-07:
The commits entered master long ago.

closed 2019-03-07 by chelmuth