genodelabs/genode #2736: intel_fb: update graphic driver to support post Intel Skylake graphic chips

alex-ab, 2018-03-29:


alex-ab, 2018-08-10:
Current shortcoming after the first testing round:
- on one specific T420 assertion is i2c contrib sources (another T420 works however)
- on X260 boot delay due to timeouts during connector discovery in contrib sources
- on one T460 and T470 eDP is just discovered as DP (so that brightness changes does not work), another T470 with close to same hardware configuration discovers it correctly as eDP

m-stein, 2018-08-13:
@skalk and I debugged the unimplemented function on my t430s but we were not able to finish it. This is the outcome so far (function calls in the intel fb contrib that are related to the error and annotations):
```
intel_gpio_setup (intel_i2c.c)
  register adapter i915 gmbus ssc   (gpio base c0000 offset 5014)
  register adapter i915 gmbus vga   (gpio base c0000 offset 5010)
  register adapter i915 gmbus panel (gpio base c0000 offset 5018)
  register adapter i915 gmbus dpc   (gpio base c0000 offset 501c)
  register adapter i915 gmbus dpb   (gpio base c0000 offset 5020)
  register adapter i915 gmbus dpd   (gpio base c0000 offset 5024)
intel_sdvo_init (init Intel Serial Digital Video Out, intel_sdvo.c)
  intel_sdvo_read_byte (during first attempt to read SDVO regs 0x0 to 0x40 via I2C, intel_sdvo.c)
    i2c_transfer (i2c-core-base.c)
      __i2c_transfer (i2c-core-base.c)
        intel_sdvo_ddc_proxy_xfer (via adap->algo->master_xfer, intel_sdvo.c)
          gmbus_xfer (via sdvo->i2c->algo->master_xfer, intel_i2c.c)
            bit_xfer (via i2c_bit_algo.master_xfer, i2c-algo-bit.c)
              bit_doAddress (i2c-algo-bit.c)
                try_address (in else case named "normal 7bit address", i2c-algo-bit.c)
                  i2c_outb (times out 1 times with retries=1, i2c-algo-bit.c)
                    sclhi (first occurence, returns "timed out", udelay arg = 5, i2c-algo-bit.c)
                      set_clock (via setscl, writes 0x1 to gpio reg 0xc5020, intel_i2c.c)
                      adap->timeout is 0 (this is also strange but setting it to 0xffff
                                          doesn't change a thing)
                      get_clock (via getscl, reads 0x200 from gpio reg 0xc5020, expects bit 0x10 = 1,
                                 never succeeds even with huge adap->timeout, intel_i2c.c)
                      return -ETIMEDOUT (-60)
  ...
  catch error of intel_sdvo_read_byte (in intel_gpio_setup, intel_sdvo.c)
    device_for_each_child (uimplemented, dummies.c)
      log "BUG: failure at .../dummies.c:101/device_for_each_child()"
```
So the problem seems to be that the "initialization finished" value for the clock of the I2C bus to the Intel SDVO doesn't show up in the GPIO reg of the i915 gmbus dpb adapter during the first attempt to read the SDVO device regs via I2C (probing).

chelmuth, 2018-08-13:
At least on docked x250/x260 and with monitor connected via DisplayPort to the dock on boot, the monitor is not enabled by default. This may be caused by a delayed initialization of the connector. After moving `_session.config_changed();` in lx_emul.cc from `finish_initialization()` to the end of `generate_report()` the known behavior is restored. I only have the feeling this may not be a valid fix.

chelmuth, 2018-08-15:
With dca21d139a18faa2ad1b8e52368eca0c195a9a77 and run/intel_fb on x260 I get two strange message snippets in the log.

```
[init -> init_dynamic -> intel_fb_drv] drm_sysfs_connector_remove not implemented
[init -> init_dynamic -> intel_fb_drv] BUG: failure at /plain/krishna/src/genode/genode_staging.git/repos/dde_linux/src/drivers/framebuffer/intel/lx_emul.cc:1251/drm_sysfs_connector_remove()!
```

and

```
[init -> init_dynamic -> intel_fb_drv] ioremap: return sub range phys 0xe0c40000 (size 4149248) to virt 0x20c40000
[init -> init_dynamic -> intel_fb_drv] Warning: schedule() called without runnable tasks
[init -> init_dynamic -> intel_fb_drv] SCHEDULE [0] prio: 3 state: 2 irq_8086:1916
[init -> init_dynamic -> intel_fb_drv] SCHEDULE [1] prio: 2 state: 2 i915_modeset
[init -> init_dynamic -> intel_fb_drv] SCHEDULE [2] prio: 2 state: 2 i915-dp
[init -> init_dynamic -> intel_fb_drv] SCHEDULE [3] prio: 2 state: 2 i915
[init -> init_dynamic -> intel_fb_drv] SCHEDULE [4] prio: 2 state: 2 system_wq
[init -> init_dynamic -> intel_fb_drv] SCHEDULE [5] prio: 2 state: 3 work_queue
[init -> init_dynamic -> intel_fb_drv] SCHEDULE [6] prio: 2 state: 2 timer
[init -> init_dynamic -> intel_fb_drv] SCHEDULE [7] prio: 0 state: 3 linux
```

The last one puzzles me most and does not happen without dca21d139a18faa2ad1b8e52368eca0c195a9a77.

nfeske, 2018-08-15:
Commit 597a46fec4ab7ebc0dcacdef0e92dd0726bcbe87 fixes the freeze when booting my x260 with an external DP panel connected. In contrast to @m-stein's commit (which triggers the problem reported by @chelmuth), my fix does not immediately call the update-mode code but merely triggers an artificial config-update signal. So the handling of hotplug events and config updates follow the same control flow.

With the additional implementation of the `device_for_each_child` and `device_unregister` functions, the driver survives the disconnect of the laptop from the docking station. However, reconnecting the laptop does not enable the external panel again. I guess that some step in the device-unregister procedure is still missing because no new connector reports are generated on any successive plug/unplug event. So maybe the driver internally still assumes that the connector stayed connected for the whole time?

@alex-ab Could you give my fixup a quick glance?

nfeske, 2018-08-20:
@cproc gave me the hint to disable the `enable_dp_mst` option in `src/drivers/framebuffer/intel/i915_params.c`. I can confirm that this fixes my docking-station problem with the x260. So I think we should disable the option by default until we have the actual need for (non-mirror) multi-head setups.

m-stein, 2018-08-21:
@cproc, @nfeske:

With current staging, the intel_fb test doesn't get stuck when connecting an external monitor  to the docking station of the Fuji Lifebook S Series that wasn't connected on bootup (in contrast to the latest internal Sculpt). Nonetheless, there is a problem with dis- and reconnecting the monitor. The behavior is for all three, VGA, DP, and DVI the same.

Intel_fb log when connecting the first time:
```
dev_dbg: adapter [DPMST] registered
ioremap: return sub range phys 0xb0440000 (size 4149248) to virt 0x20440000
[drm_wait_one_vblank] *WARN* vblank wait timed out on crtc 0
```
Now, the Monitor works just fine.
Intel_fb log when disconnecting the first time:
```
dev_dbg: adapter [DPMST] unregistered
ioremap: return sub range phys 0xb0040000 (size 4149248) to virt 0x20040000
Error: set config failed ret=-22 refcnt before=2 after=2
Warning: framebuffer refcount 2
```
Afterwards, connecting and disconnecting the monitor shows no effect anymore and leaves the monitor without signal.

If the monitor is connected on bootup, it's the same but with the first log lines appearing during bootup.
I've tried setting  ``enable_dp_mst`` in ``intel/drivers/gpu/drm/i915/i915_params.h`` to ``false`` as suggested by Norman but it didn't changed a thing.

I'll have a deeper look at it now.

cproc, 2018-08-21:
@m-stein: does connecting the monitor for the first time add new connectors in the connector report of the intel_fb test? If it does, DP MST support might still be enabled for some reason. In that case there would be driver cleanup problems when a connector is removed in the driver. For example, the connector report would still list the connector with its modes, but setting a mode would fail (with an error message like your "set config failed" message). Also, when the mode is set for the new connector, a reference count is incremented for the connector object in the driver, but not decremented when the monitor is removed and for that reason the connector object would not get destroyed and connecting the monitor again would create a new connector object and when doing that a few times, a "Found no crtc for the connector" error message would appear.

m-stein, 2018-08-21:
There are always three connectors in the Intel FB report:

eDP-1
Is always "connected".

HDMI-A-1
Becomes "connected" whenever laptop HDMI gets connected (independent of the DS at all)

DP-1
Becomes "connected" if booted **without** DS and then laptop VGA gets connected (independent of the later availability of the DS)
Becomes "connected" if booted **with** DS and then DS VGA/DVI/DP get connected

So there seems to be a problem with changing the routing of DP-1 (to DS or laptop VGA) that was set during boot. With Linux, the pattern is "without DS => DP-1 routed to laptop VGA, with DS => DP-1 routed to DS".

m-stein, 2018-08-28:
On Linux, the drm is not called when the docking station appears/disappears but when I compile the kernel without ACPI, the symptoms are exactly the same as with the Genode setup.

chelmuth, 2018-08-29:
This seems to substantiate the guess that the driver does not control the docking graphics-wise. Did you test the Genode scenario with a running acpica component already? Maybe we're lucky and proper ACPI reporting/setup is sufficient to enable the magic in the platform.

commit 07bb227d (2018-08-08): dde_linux: update intel_fb to 4.16.3

Fixes #2736
commit eb62d9cc (2018-08-28): dde_linux: update intel_fb to 4.16.3

Fixes #2736