genodelabs/genode #2738: nic_router: DHCP handshake fails on some WiFi networks

MartijnVerschoor, 2018-03-30:
In configurations where the nic_router uplink is routed to the wifi_drv, on some WiFi networks the DHCP handshake does not come to completion. Instead of sending the DHCP acknowledgement, the DHCP server on the WiFi network issues an ARP request, which is left unresolved:

```
[init -> nic_router] (uplink <- router) ETH bc:a8:a6:7d:c2:65 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP bc:a8:a6:7d:c2:65 > 0.0.0.0 cmd 1
[init -> nic_router] (router <- uplink) ETH 08:76:ff:42:03:7d > ff:ff:ff:ff:ff:ff IPV4 192.168.1.254 > 255.255.255.255 UDP 67 > 68 DHCP bc:a8:a6:7d:c2:65 > 0.0.0.0 cmd 2
[init -> nic_router] (uplink <- router) ETH bc:a8:a6:7d:c2:65 > ff:ff:ff:ff:ff:ff IPV4 192.168.1.209 > 255.255.255.255 UDP 68 > 67 DHCP bc:a8:a6:7d:c2:65 > 0.0.0.0 cmd 1
[init -> nic_router] (router <- uplink) ETH 08:76:ff:42:03:7d > ff:ff:ff:ff:ff:ff ARP 08:76:ff:42:03:7d 192.168.1.254 > 00:00:00:00:00:00 192.168.1.209 cmd 1
```

On some WiFi networks it works fine:

```
[init -> nic_router] (uplink <- router) ETH bc:a8:a6:7d:c2:65 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP bc:a8:a6:7d:c2:65 > 0.0.0.0 cmd 1
[init -> nic_router] (router <- uplink) ETH c4:e9:84:c8:69:5b > ff:ff:ff:ff:ff:ff IPV4 192.168.0.1 > 255.255.255.255 UDP 67 > 68 DHCP bc:a8:a6:7d:c2:65 > 0.0.0.0 cmd 2
[init -> nic_router] (uplink <- router) ETH bc:a8:a6:7d:c2:65 > ff:ff:ff:ff:ff:ff IPV4 192.168.0.105 > 255.255.255.255 UDP 68 > 67 DHCP bc:a8:a6:7d:c2:65 > 0.0.0.0 cmd 1
[init -> nic_router] (router <- uplink) ETH c4:e9:84:c8:69:5b > bc:a8:a6:7d:c2:65 IPV4 192.168.0.1 > 192.168.0.105 UDP 67 > 68 DHCP bc:a8:a6:7d:c2:65 > 0.0.0.0 cmd 2
```

I'd like to provide more detailed information, but I'm not sure how to collect it.

nic_router config:
```
<config verbose="yes">
	<domain name="uplink">
		<nat domain="vm2" tcp-ports="1000" udp-ports="1000"/>
	</domain>
	<policy label_prefix="starter -> vm2 -> nic_relay" domain="vm2" />
	<domain name="vm2" interface="192.168.2.1/24">
		<dhcp-server ip_first="192.168.2.2"
		             ip_last="192.168.2.250"
		             ip_lease_time_sec="3600"
		             dns_server="8.8.8.8"/>
		<tcp dst="0.0.0.0/0"> <permit-any domain="uplink" /> </tcp>
		<udp dst="0.0.0.0/0"> <permit-any domain="uplink" /> </udp>
	</domain>
</config>
```

chelmuth, 2018-04-03:
Could you please try to change nic_router's implementation in a way that it does not use the *future* IP address as sender address during DHCP REQUEST? I don't have a poitner into the RFC but the [Wikipedia article](https://en.wikipedia.org/wiki/Dynamic_Host_Configuration_Protocol#DHCP_request)  specifies `IP: source=0.0.0.0 destination=255.255.255.255` with the exception of unicast destination if the DNS server is known upfront.

m-stein, 2018-04-04:
I have not found so far a hint in the RFCs but changed the implementation according to the Wikipedia article and it works with the virtualbox_nic_router.run script:
abeca34 nic_router: do DHCP requests without source IP

@MartijnVerschoor Could you give it a try?

chelmuth, 2018-04-05:
The information is hidden in Section [4.1 Constructing and sending DHCP messages
](https://tools.ietf.org/html/rfc2131#section-4.1) of [RFC 2131](https://tools.ietf.org/html/rfc2131)

> DHCP messages broadcast by a client prior to that client obtaining its IP address must have the source address field in the IP header set to 0.

and could also be interpreted in either way.

MartijnVerschoor, 2018-04-05:
Hi Martin, I gave abeca348491898238ef4e7fa68503eec586afc97 a try, pretty much on top of 18.02.

Here are my findings:
- I can still connect to the WiFi network that worked before ;-)
- abeca348491898238ef4e7fa68503eec586afc97 solves the issue for one network.
- On another network the DHCP handshake is still troubled. The DHCP handshake gets one step further but eventually fails over a missing DHCP option...?:

```
[init -> nic_router] (uplink <- router) ETH bc:a8:a6:7d:c2:65 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP bc:a8:a6:7d:c2:65 > 0.0.0.0 cmd 1
[init -> nic_router] (uplink <- router) ETH bc:a8:a6:7d:c2:65 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP bc:a8:a6:7d:c2:65 > 0.0.0.0 cmd 1
[init -> nic_router] (router <- uplink) ETH b2:fd:61:61:05:64 > bc:a8:a6:7d:c2:65 IPV4 172.20.10.1 > 172.20.10.4 UDP 67 > 68 DHCP bc:a8:a6:7d:c2:65 > 172.20.10.1 cmd 2
[init -> nic_router] (uplink <- router) ETH bc:a8:a6:7d:c2:65 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP bc:a8:a6:7d:c2:65 > 0.0.0.0 cmd 1
[init -> nic_router] (router <- uplink) ETH b2:fd:61:61:05:64 > bc:a8:a6:7d:c2:65 IPV4 172.20.10.1 > 172.20.10.4 UDP 67 > 68 DHCP bc:a8:a6:7d:c2:65 > 172.20.10.1 cmd 2
[init -> nic_router] Warning: Had to prune the state timeout of DHCP client
[init -> nic_router] (uplink) Drop packet: DHCP client misses DHCP option

[...] Start VM [...]

[init -> nic_router] (router <- vm2) ETH 02:02:02:02:02:00 > 01:00:5e:00:00:fb IPV4 192.168.2.2 > 224.0.0.251 UDP 5353 > 5353 
[init -> nic_router] Using UDP rule: 0.0.0.0/0 requests to uplink
[init -> nic_router] (vm2) Drop packet: target domain has yet no IP config
```
Perhaps the latter is another issue.

As far as I can tell, abeca348491898238ef4e7fa68503eec586afc97 is an improvement. I agree with @chelmuth that it is more inline with the RFC. It makes sense to not use an offered IP before it is requested and acknowledged.

chelmuth, 2018-04-05:
@MartijnVerschoor this looks strange. could it be caused by a merge error?

For background, the exception is thrown only in the case a DHCP option is requested that's not in the packet received from the server. nic_router seems to check for `Ip_lease_time` (which seems to work due to the warning), `Subnet_mask`, and `Router_ipv4`. (The current implementation probes for `Dns_server_ipv4` too, but catches the exception). So for identifying the option causing trouble, you may instrument by
``` diff
-       catch (Dhcp_packet::Option_not_found) {
-               throw Drop_packet_inform("DHCP client misses DHCP option");
+       catch (Dhcp_packet::Option_not_found e) {
+               throw Drop_packet_inform("DHCP client misses DHCP option ", (unsigned)e.code);
```
in *dhcp_client.cc*.

MartijnVerschoor, 2018-04-05:
I can pin it to option nr 3: `ROUTER`:
```
 (uplink) Drop packet: DHCP client misses DHCP option 3
```

I am not aware of merge conflicts. I am currently running an integrated scenario. For this scenario we have a few commits on top of 18.02 but none should impact the nic router. I double checked the paths:
- `repos/os/src/server/nic_router`
- `os/include/net`

They are clean 18.02 + abeca348491898238ef4e7fa68503eec586afc97

chelmuth, 2018-04-05:
So the questions is: Is there no router/gateway advertised (and we should cope with that fact by using only the local network segment) or do we have a parsing error? Is this a local test network without configured routing/uplink?

MartijnVerschoor, 2018-04-05:
The particular access point serves a WAN. I checked with Wireshark on a Linux client that the associated DHCP server advertises the corresponding IPv4 router in the DHCP offer in an option 3 subpacket. 

However, I don't see the option in the offer to the Genode client...

I inserted a nic_dump in between the nic_router and the wifi_drv to get a bit more detail. I configured nic_dump DHCP to be COMPREHENSIVE. This is the DHCP exchange:

```
[init -> nic_router] (uplink <- router) ETH bc:a8:a6:7d:c2:65 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP bc:a8:a6:7d:c2:65 > 0.0.0.0 cmd 1

[init -> nic_dump] (wifi_drv <- nic_router) ETH bc:a8:a6:7d:c2:65 > ff:ff:ff:ff:ff:ff  IPV4 0.0.0.0 > 255.255.255.255  UDP 68 > 67  DHCP op 1 htyp 1 hlen 6 hps 0 xid 305419896 sec 0 flg 0x0 ci 0.0.0.0 yi 0 .0.0.0 si 0.0.0.0 gi 0.0.0.0 ch bc:a8:a6:7d:c2:65 srv file  mag 1669485411 opt 53:1:01 61:7:01bca8a67dc265 57:2:03d6 time 23957 ms ( 10000 ms)

[init -> nic_dump] (nic_router <- wifi_drv) ETH b2:fd:61:61:05:64 > bc:a8:a6:7d:c2:65  IPV4 172.20.10.1 > 172.20.10.4  UDP 67 > 68  DHCP op 2 htyp 1 hlen 6 hps 0 xid 305419896 sec 0 flg 0x0 ci 0.0.0.0 yi 1 72.20.10.4 si 172.20.10.1 gi 0.0.0.0 ch bc:a8:a6:7d:c2:65 srv Woz file  mag 1669485411 opt 53:1:02 54:4:ac140a01 51:4:00014e20 time 24225 ms ( 268 ms)

[init -> nic_router] (router <- uplink) ETH b2:fd:61:61:05:64 > bc:a8:a6:7d:c2:65 IPV4 172.20.10.1 > 172.20.10.4 UDP 67 > 68 DHCP bc:a8:a6:7d:c2:65 > 172.20.10.1 cmd 2

[init -> nic_router] (uplink <- router) ETH bc:a8:a6:7d:c2:65 > ff:ff:ff:ff:ff:ff IPV4 0.0.0.0 > 255.255.255.255 UDP 68 > 67 DHCP bc:a8:a6:7d:c2:65 > 0.0.0.0 cmd 1

[init -> nic_dump] (wifi_drv <- nic_router) ETH bc:a8:a6:7d:c2:65 > ff:ff:ff:ff:ff:ff  IPV4 0.0.0.0 > 255.255.255.255  UDP 68 > 67  DHCP op 1 htyp 1 hlen 6 hps 0 xid 305419896 sec 0 flg 0x0 ci 0.0.0.0 yi 0 .0.0.0 si 0.0.0.0 gi 0.0.0.0 ch bc:a8:a6:7d:c2:65 srv  file  mag 1669485411 opt 53:1:03 61:7:01bca8a67dc265 57:2:03d6 50:4:ac140a04 54:4:ac140a01 time 24227 ms ( 2 ms)

[init -> nic_dump] (nic_router <- wifi_drv) ETH b2:fd:61:61:05:64 > bc:a8:a6:7d:c2:65  IPV4 172.20.10.1 > 172.20.10.4  UDP 67 > 68  DHCP op 2 htyp 1 hlen 6 hps 0 xid 305419896 sec 0 flg 0x0 ci 0.0.0.0 yi 1 72.20.10.4 si 172.20.10.1 gi 0.0.0.0 ch bc:a8:a6:7d:c2:65 srv Woz file  mag 1669485411 opt 53:1:05 54:4:ac140a01 51:4:00014e20 time 24289 ms ( 62 ms)

[init -> nic_router] (router <- uplink) ETH b2:fd:61:61:05:64 > bc:a8:a6:7d:c2:65 IPV4 172.20.10.1 > 172.20.10.4 UDP 67 > 68 DHCP bc:a8:a6:7d:c2:65 > 172.20.10.1 cmd 2

[init -> nic_router] Warning: Had to prune the state timeout of DHCP client
[init -> nic_router] (uplink) Drop packet: DHCP client misses DHCP option
```

chelmuth, 2018-04-06:
The log is not easy to decipher but looks okay except for the fact of the missing router option. Could you PM me pcap capture files of the DHCP packets for a working Linux client?

chelmuth, 2018-04-06:
Hm, my Linux is sending a specific option what info it wants to get from the DHCP server (on rerequest at least), which is documented in the RFC section 9.8. Parameter Request List. Maybe this should be added to our REQUEST/DISCOVER for interoperability.

m-stein, 2018-04-06:
I'll implement it.

chelmuth, 2018-04-06:
Thanks @MartijnVerschoor! For me it looks like your router's DHCP server does send the router option only if requested via option 55. @m-stein will implement the feature and count on you to give it a spin after completion.

MartijnVerschoor, 2018-04-06:
Sure, thanks!

m-stein, 2018-04-06:
I added some 5 commits, 3 that improve nic_dump and 2 that enable parameter request lists in the nic_router:
6c5eb0d nic_router: use DHCP parameter request list
4f90be1 net: utilities for DHCP parameter-request lists
ad925db nic_dump: add config XML schema file
7d1742f nic_dump: default attribute
a337d1b nic_dump: make protocols configurable

MartijnVerschoor, 2018-04-09:
Thanks Martin! 
I gave it a try but unfortunately cherry-picking these 5 commits yields a few merge conflicts. When resolved with the recursive theirs strategy, the code won't compile; an indication that I'm missing some commits.

Could you give me link to the branch your'e working on?

chelmuth, 2018-04-09:
The branch is m-stein/merge_to_staging.

MartijnVerschoor, 2018-04-10:
Thanks @m-stein, the DHCP parameter-request list commits solves the issue. Below a log snippet of nic_dump showing the dhcp exchange:

```
[init -> runtime -> nic_dump] (nic_drv <- vfs) ETH 94:65:9c:9b:1d:d8 > ff:ff:ff:ff:ff:ff  IPV4 0.0.0.0 > 255.255.255.255  UDP 68 > 67  DHCP op 1 htyp 1 hlen 6 hps 0 xid 305419896 sec 0 flg 0x0 ci 0.0.0.0 yi 0.0.0.0 s
[init -> runtime -> nic_dump] i 0.0.0.0 gi 0.0.0.0 ch 94:65:9c:9b:1d:d8 srv  file  mag 1669485411 opt 53:1:01 55:6:353633060103 61:7:0194659c9b1dd8 57:2:03d6 time 31004 ms (Δ 10000 ms)
[init -> runtime -> nic_dump] (vfs <- nic_drv) ETH b2:fd:61:61:05:64 > 94:65:9c:9b:1d:d8  IPV4 172.20.10.1 > 172.20.10.2  UDP 67 > 68  DHCP op 2 htyp 1 hlen 6 hps 0 xid 305419896 sec 0 flg 0x0 ci 0.0.0.0 yi 172.20.10
[init -> runtime -> nic_dump] .2 si 172.20.10.1 gi 0.0.0.0 ch 94:65:9c:9b:1d:d8 srv Woz file  mag 1669485411 opt 53:1:02 54:4:ac140a01 51:4:00014e20 6:4:ac140a01 1:4:fffffff0 3:4:ac140a01 time 31062 ms (Δ 58 ms)
[init -> runtime -> nic_dump] (nic_drv <- vfs) ETH 94:65:9c:9b:1d:d8 > ff:ff:ff:ff:ff:ff  IPV4 0.0.0.0 > 255.255.255.255  UDP 68 > 67  DHCP op 1 htyp 1 hlen 6 hps 0 xid 305419896 sec 0 flg 0x0 ci 0.0.0.0 yi 0.0.0.0 s
[init -> runtime -> nic_dump] i 0.0.0.0 gi 0.0.0.0 ch 94:65:9c:9b:1d:d8 srv  file  mag 1669485411 opt 53:1:03 55:6:353633060103 61:7:0194659c9b1dd8 57:2:03d6 50:4:ac140a02 54:4:ac140a01 time 31077 ms (Δ 15 ms)
[init -> runtime -> nic_dump] (vfs <- nic_drv) ETH b2:fd:61:61:05:64 > 94:65:9c:9b:1d:d8  IPV4 172.20.10.1 > 172.20.10.2  UDP 67 > 68  DHCP op 2 htyp 1 hlen 6 hps 0 xid 305419896 sec 0 flg 0x0 ci 0.0.0.0 yi 172.20.10
[init -> runtime -> nic_dump] .2 si 172.20.10.1 gi 0.0.0.0 ch 94:65:9c:9b:1d:d8 srv Woz file  mag 1669485411 opt 53:1:05 54:4:ac140a01 51:4:00014e20 6:4:ac140a01 1:4:fffffff0 3:4:ac140a01 time 31149 ms (Δ 72 ms)
[init -> runtime -> nic_router] Warning: Had to prune the state timeout of DHCP client
[init -> runtime -> nic_router] [uplink] IP config: interface 172.20.10.2/28, gateway 172.20.10.1, DNS server 172.20.10.1
```

closed 2018-04-10 by chelmuth