genodelabs/genode #2739: Expose executable memory allocator in Libc ABI

ehmry, 2018-03-30:
Applications that generate and execute code at runtime need memory that that they can modify and execute from. For at least NOVA and base-hw it is not the case that arbitrary memory returned by malloc will be executable, and for this reason the DOSbox dynamic recompiler no longer works with these kernels.

The libc currently contains a malloc like allocator for executable memory at `repos/libports/src/lib/libc/libc_mem_alloc.cc`, why not expose C symbols for this allocator as an extension to the libc ABI? Modifying applications to use a `malloc_exec` should be trivial because the required amount of executable memory  is small. For example, DOSbox only requires a single executable allocation in its virtual CPU to work. Using an explicit executable malloc rather than overriding `malloc` for an entire libc application is better anyway because it adds more protection against attacks that might escape a JIT compiler.

ehmry, 2018-03-30:
Patch at 199d548722505957da1bb0f30bb766cd7213536a.

chelmuth, 2018-04-04:
I must admit that the new interface does not make me happy for two reasons. First, the introduction of a new class of memory allocations carries the risk of mixing up the different allocators, ie., freeing non-exec at the exec allocator. Second, the libc comes with a supported interface for executable anonymous-memory allocations via `mmap(..., PROT_EXEC, MAP_ANONYMOUS,...)` that obsoletes the hacky `mprotect` on heap allocations. Do you see a chance to adapt dosbox to mmap instead of using your custom interface with reasonable effort?

ehmry, 2018-04-04:
Yes, mmap works and is more portable, thanks.

closed 2018-04-04 by ehmry