genodelabs/genode #2741: regression: usb_drv breaks platform_drv on AMD hardware

NobodyIII, 2018-04-02:
As I mentioned on the mailing list, usb_drv no longer works on my desktop computer, which has an AMD 9xx chipset and an FX-8320 CPU.

Here are the key errors I see when running the usb_hid scenario:
[init -> platform_drv] Warning: Out_of_caps during Device_component construction
[init -> platform_drv] Warning: Out_of_caps during Device_component construction
[ 0] sys_assign_pci: Invalid Hint (0x200)

Please note that I see the sys_assign_pci error in both the usb_hid and ahci_bench scenarios, but the Out_of_caps error only appears in usb_hid. The framebuffer scenario (which also uses PCI) works just fine.

These errors don't go away when I give platform_drv 2000 CAPs (as opposed to 800).

It seems that platform_drv is probably part of the issue, given that it's the component that fails. At the very least, it shouldn't hang the system (as I've seen with a modified version of run/demo) due to bugs in other drivers.

NobodyIII, 2018-04-02:
It looks like usb_drv gets the Out_of_caps warning in QEMU, but it still works just fine, so I think it's safe to ignore here.

I modified the usb_hid scenario to use the bios handoff quirk, which is enabled in the driver by default, but disabled in the usb_hid scenario. I also disabled xHCI, because it seemed like it might be causing problems.

After all of this, the scenario showed no errors, and only the Out_of_caps warning, which seems to be safe. When I tested the AHCI driver with ahci_bench, I found that its log output was mostly normal, but incomplete.

It seems like the problem really comes down to interrupts not working. The only driver that seems to work is the VESA framebuffer driver, which doesn't need to receive interrupts.

NobodyIII, 2019-05-05:
This problem doesn't show up on my AMD Ryzen system, so I'm not planning to pursue this further. Should I close the issue?

chelmuth, 2019-05-06:
As you are the original reporter you may close issues that lost relevance to you.

closed 2019-11-19 by alex-ab