genodelabs/genode #2742: Vfs constructor cleanup

ehmry, 2018-04-02:
As mentioned in #1934 and #2701.

The current VFS constructor has become quite large, it carries a `Genode::Env`, `Genode::Allocator`, an XML configuration, an `Io_response_handler` for callbacks` and a root `Vfs::File_system`. It is conceivable that more objects may be passed in the future, but adding arguments to the constructor breaks all VFS plugins, including those hosted in external repositories.

I suggest creating a `Vfs::Env` object to carry the current and future arguments, aside from per-file-system configuration. I had considered other names like `Core` or `Root`, but I think `Env` would be most consistent.

The only non-trivial part would be the `Io_response_handler`. The current `Io_response_handler` handles both I/O response and watch responses, I would create a new `Watch_response_handler` object and revert the `Io_response_handler` to handle I/O only. Both would be accessible from `Vfs::Env`.

I would propose to add a lazily constructed timer source for timeouts. In my own experiments I've seen that TCP plugins work a bit faster when they use the timeout scheduling facility rather than rely on simple signal handling, but this can wait.

nfeske, 2018-04-03:
Everything you propose above sounds very good to me.

chelmuth, 2018-04-03:
Your proposal makes perfect sense. Go for it.

ehmry, 2018-04-03:
Fixed at 5f1706e0e68070c30e9eef1f0b462aec673c21bb.

nfeske, 2018-04-04:
Wow, that's a huge patch. Thanks for this very thorough work! While reviewing it, the following thoughts crossed my mind:

* With the addition of the `Vfs_trivial_env`, the `Vfs::Env` interface becomes visible at the designated client-side API, which I would prefer to avoid. Let us better integrate this class into the existing `Root_directory`. How about privately inheriting the response handlers as well as `Vfs::Env` from `Root_directory`? BTW, this header uses the Genode namespace. So there is no need to use the `Genode::` prefix. 

* You removed the distinction between the root and non-root constructors of the `Dir_file_system` by the check `this == env.root_dir`, which is quite clever. However, when using the `Dir_file_system` as utility for VFS plugins (like I did for the ttf plugin - expecting that this convenient pattern will be copied), it might be preferable to make this distinction more explicit, e.g., by letting `_vfs_root` depend on the type of the config node passed to the constructor (via `config.has_type("dir")`). When used as a utility from a vfs plugin, the plugin passes a custom config to the constructor anyway. With this approach, the plugin can decide which "mode" of the `Dir_file_system` it wants to use. In contrast, with your check, the plugin would need to craft a custom `Vfs::Env` to operate the `Dir_file_system` in root mode.

nfeske, 2018-04-05:
My long-term plan is to move `gems/vfs.h` to the os repository. But I first want to let it settle and cover more use cases (like writing files, watching changes). So I leave it at gems as our experimentation ground. The `Root_directory` is particularly unfinished because it does not use any io-response feature. It is probably too early to move it to os? I would feel better to keep a copy of the code at both places for now, instead of extending the prominently visible os headers. However, if you come up with a third user, it may be time to move it to os. ;-)

nfeske, 2018-04-10:
Merged to staging

nfeske, 2018-04-10:
With these commits and the COW commits merged, I get a lot of (false-positive) warnings, e.g., when executing run/depot_deploy. @ehmry Was I too eager with merging the COW commits?

ehmry, 2018-04-12:
Yes, thats too many warnings, I can make a patch to remove the warnings, but I think it shows that `fs_rom` is going to make a bunch of synchronous calls before returning new sessions, so I would be willing to produce a simple static `fs_rom` that can be amended later with a cache of read-only ROM dataspaces.

nfeske, 2018-04-12:
Unfortunately, the commit breaks several scenarios on staging. I.e., noux_shell_script.run (actually, seemingly all run scripts that spawn bash with --login) does not start anymore. While browsing the commit again, I could not spot an obvious reason. @ehmry Could you also have a look?

nfeske, 2018-04-12:
The problem can be triggered by running noux_shell_script.run on base-linux. It hangs on fstat within core. In the backtrace, I can see that noux performs a `Noux::Vfs_io_channel::_sync` and hangs at
```
		while (_fh->fs().complete_sync(_fh) == Vfs::File_io_service::SYNC_QUEUED)
			vfs_io_waiter.wait_for_io();
```

nfeske, 2018-04-12:
The problem seem to be my naive use of `Simple_env` in noux. This cannot work because the `Simple_env` does not implement an io-response handler. Silly me.

nfeske, 2018-04-12:
Yeah, in particular the circular dependency between the root directory and the env won't win a beauty contest. But I don't see an simple way to avoid it right now.

nfeske, 2018-04-12:
Commit https://github.com/genodelabs/genode/commit/8fdb96eacd54b80d60f113988fa16ac092a06ed6 fixes the noux problem.

nfeske, 2018-04-12:
As another problem on staging, the commit https://github.com/genodelabs/genode/commit/5b31d532a03ea42150efa3799f1614598cda236d apparently breaks `depot_download.run`. I have reverted this specific commit as an interim solution. But I suspect that the subsequent vfs-cow commit won't work as intended now. @ehmry may you take a closer look? ;-)

nfeske, 2018-04-12:
> Yes, thats too many warnings, I can make a patch to remove the warnings

Even with the static_fs_rom in place, I would prefer to avoid those warnings. I think, watching static files isn't a problem to warn about. It's just boring.

ehmry, 2018-04-12:
Silenced at acb51775970e5223d3230793f8cccde69835a952.

ehmry, 2018-04-13:
I've fixed the depot_download scenario at 0abfb8b63270161c5feb9030e7d885eab99fc5a8. I believe the for some time now the `VFS::Dir_file_system` has allowed empty virtual directories to be opened without any backing, but the libc hid this problem by doing a `stat` check first.

nfeske, 2018-04-13:
Thanks a lot. I reverted the reverted commit and added your fix to staging.

nfeske, 2018-04-13:
Added the silencing commit acb51775970e5223d3230793f8cccde69835a952 as fixup to the original commit to staging.

ehmry, 2018-04-13:
@chelmuth fixup for cli_monitor and vfs_stress at 17e5bb4ee9135bd2bcc9db073837e977a3f78e43.

commit 5127db99 (2018-04-10): VFS: construct file-systems using Vfs::Env object

Reduce the size and forward compatibility of VFS file-system
constructors by passing an object holding accessors for 'Genode::Env',
'Genode::Allocator', response handlers, and the root file-system.

Fix #2742
commit b9c34b6d (2018-04-10): VFS: construct file-systems using Vfs::Env object

Reduce the size and forward compatibility of VFS file-system
constructors by passing an object holding accessors for 'Genode::Env',
'Genode::Allocator', response handlers, and the root file-system.

Fix #2742
commit 82a683ec (2018-04-19): VFS: construct file-systems using Vfs::Env object

Reduce the size and forward compatibility of VFS file-system
constructors by passing an object holding accessors for 'Genode::Env',
'Genode::Allocator', response handlers, and the root file-system.

Fix #2742
closed 2018-04-19 by chelmuth