genodelabs/genode #2744: Resize nit_fb to requested client mode

ehmry, 2018-04-03:
I would like to change the resize of a framebuffer session during the lifetime of a component by opening and closing the session with different mode session argument as mentioned in  #2728. Nit_fb however uses a static framebuffer which the client has no control over. I have a patch to resize the nit_fb framebuffer and view at the request of the client, however the client mode will override the current size set by nitpicker and the size set in the current configuration.

ehmry, 2018-04-03:
Patch at 691ec94cec7b71e0e9ca91539fe7683dd13ddd44.

nfeske, 2018-04-04:
Good idea. It would be nice to restrict this feature to the case where `width` and `height` are not defined in the configuration though. So the configuration has ultimate power over the mode whenever needed. In contrast, if `initial_width` and `initial_height` are specified in the config, responding to the client-provided mode is fine.

ehmry, 2018-04-13:
I'm closing this issue because I've decided to use a Nitpicker session rather than Framebuffer. With Nitpicker I can allocate a framebuffer as large as I need and then resize within that area. I will leave the branch around however if someone else is has a use case for such a feature.

ehmry, 2018-06-11:
I've changed my mind and decide that this is a feature worth having. I've adjusted the patch to prevent the client from changing the current Framebuffer mode if the configuration of nit_fb has any explicit dimensions. Fixed at f0aa2caa2a949de5d708861c8690985babf1eb82.

nfeske, 2019-03-17:
With the move towards the use of nitpicker directly (see #3196) this issue becomes obsolete.

@ehmry would you be fine with closing it?

ehmry, 2019-03-18:
Yes, I just use Nitpicker now.

closed 2019-03-18 by ehmry