genodelabs/genode #2745: Copy-on-write VFS plugin

ehmry, 2018-04-04:
With #2701 cleared it is time for the anticipated copy-on-write VFS plugin. This is a general purpose plugin that has many uses. An immediate use-case would be to define an static, early file-system in a boot image that seeds a mutable file-system in local, persistent storage.

ehmry, 2018-04-04:
I have a trivial implementation on my **issue2745_vfs_cow** branch. I say trivial because files are copied on open, not on write, and because the union directory behavior is roughly the same as the `Dir_file_system`. That is, the number of entries in a directory is the sum of a copied directory and the original, with the copied entries stacked ahead of the original.

I have a test at `run/fs_report_cow` that tests same use-case as `run/fs_report`, but I would like a more robust test to be confident in the COW behavior. Are there any suggestions for a use-cases to test?

closed 2018-04-19 by chelmuth

ehmry, 2018-08-10:
@nfeske cow plugin removed with 797792570345b9d2bfd35bf1e568cb1d5cdd34fd.

nfeske, 2018-08-10:
Thanks @ehmry. I merged the removal commit to staging.

commit fdfdf95c (2018-04-10): Copy-on-write VFS plugin

VFS plugin to replicate from one file-system to another. Can be used to
seed a mutable file-system with an immutable file-system. The plugin is
configure with two paths, a read-only path, and a read-write path. This
is an initial implementation that copies files on open. It is not
optimized to perform actual copy-on-write, but the result is the same.

<vfs>
	<dir name="immutable"> ... </dir>
	<dir name="mutable">   ... </dir>
	<dir name="cow">
		<cow ro="/immutable" rw="/mutable"/>
	</dir>
</vfs>

Fix #2745
commit a26b7bc5 (2018-04-10): Copy-on-write VFS plugin

VFS plugin to replicate from one file-system to another. Can be used to
seed a mutable file-system with an immutable file-system. The plugin is
configure with two paths, a read-only path, and a read-write path. This
is an initial implementation that copies files on open. It is not
optimized to perform actual copy-on-write, but the result is the same.

<vfs>
	<dir name="immutable"> ... </dir>
	<dir name="mutable">   ... </dir>
	<dir name="cow">
		<cow ro="/immutable" rw="/mutable"/>
	</dir>
</vfs>

Fix #2745
commit c1ff581f (2018-04-19): Copy-on-write VFS plugin

VFS plugin to replicate from one file-system to another. Can be used to
seed a mutable file-system with an immutable file-system. The plugin is
configure with two paths, a read-only path, and a read-write path. This
is an initial implementation that copies files on open. It is not
optimized to perform actual copy-on-write, but the result is the same.

<vfs>
	<dir name="immutable"> ... </dir>
	<dir name="mutable">   ... </dir>
	<dir name="cow">
		<cow ro="/immutable" rw="/mutable"/>
	</dir>
</vfs>

Fix #2745