genodelabs/genode #2751: net: safer access to packet data

m-stein, 2018-04-10:


m-stein, 2018-04-10:
Fixed:
48ac695 nic_router: use default Ethernet_frame constructor
0ca214d ping: use default Ethernet_frame constructor
31bdbef net/udp.h: safer access to data
6d9fa61 net/ethernet.h: safer access to data
59cfa8e net/ipv4.h: safer access to data

@nfeske I hope this first step complies with your idea of how to rework the net lib?

There exist future plans to further adapt the net lib:
* get rid of the non-standardized c++ bit indicators like in ipv4.h: ``unsigned _header_length : 4;``
* consider Ipv4 options
* re-think concept of size guards (they also may be used in reinterpret_data analogous to their use in construct_at_data)

m-stein, 2018-04-12:
Added 3 minor fixes that came up while fixing this issue:
31be3c1 test/lwip clients: exit with error code
7fc9dbe test/lwip/udp/client: handle missing attributes
4bf193c nic_router: more descriptive errors in DHCP server
Previous history:
c7aa72f nic_router: use default Ethernet_frame constructor
14d9b32 ping: use default Ethernet_frame constructor
6d7f895 net/udp.h: safer access to data
87e9135 net/ethernet.h: safer access to data
fa4ecd0 net/ipv4.h: safer access to data

chelmuth, 2018-04-16:
IMO fa4ecd0, 87e9135, 6d7f895, 14d9b32, and c7aa72f should be merged into one commit as all address the same issue. After that and the use of the original `data()` naming, I'm ready to merge the commits into staging.

m-stein, 2018-04-17:
@chelmuth I've updated the commits according to the suggestions:
ca8086d nic_router: use default Ethernet_frame constructor
15b9d72 ping: use default Ethernet_frame constructor
421eeb9 net: safer access to packet data

m-stein, 2018-04-17:
Sorry, this is the final commit that contains all five original commits:
5353d86 net: safer access to packet data

chelmuth, 2018-04-25:
Should be merged to staging now, please check.

m-stein, 2018-04-25:
Yes, as far as I can see it's all merged. Thanks!

commit 8ee64b13 (2018-04-25): net: safer access to packet data

Replace packet method 'T *data' by the new methods 'T &reinterpret_data'
for parsing or modifying existing sub-protocol packets and 'T
&construct_at_data' for composing a new sub-protocol packet. This has
the advantage that, when composing a new packet, the default constructor
that zero-fills the packet is always called first.

Fixes #2751
closed 2018-05-03 by chelmuth