genodelabs/genode #2755: init requires double quotas when running under sequence

NobodyIII, 2018-04-10:
The sequence component automatically assigns its children half of its available RAM and CAP quotas, with the expectation that if they need more of either one, they will request it. This doesn't work with init, which never requests quota upgrades. Because of this, init requires double quotas when running under sequence.

The ideal solution would be for init to request quota upgrades when needed. Is there a good reason not to do this?

nfeske, 2019-03-17:
This issue is stale for a long time now. For the current use cases of the sequence tool that I'm aware of, the current built-in policy does not stand in the way. AFAIK there is no plan of anyone to work on this issue.

@NobodyIII would you be fine to close it?

NobodyIII, 2019-03-18:
I suppose we can close this. I'm planning on fixing it by making init able
to request quota upgrades, but I'm not sure when I'll get around to that.

closed 2019-03-18 by nfeske