genodelabs/genode #2759: Turn VFS into a shared library + depot src archive

nfeske, 2018-04-11:
Since the vfs library is increasingly used by native components without a dependency on the libc (an example is depot_query), is would be sensible to turn the static library into a dynamic one and to create a dedicated source and api packages. This would also eliminate transitive API dependencies of vfs-using components from `block_session`, `rtc_session` and the likes.

ehmry, 2018-04-23:
I volunteer.

ehmry, 2018-04-23:
Patch at 9ec20bc6d6bde631a80166bebd76c3cfdae30683, though I probably missed a few things.

nfeske, 2018-04-23:
Thanks for this comprehensive patch, @ehmry!

I am wondering, why do we need to execute static constructors when using the VFS library?

How do you feel about introducing an ABI for the api/vfs archive? This would relieve VFS users from building the shared lib in each package.

ehmry, 2018-04-23:
@nfeske I added a symbols file for the VFS library and API package, is this enough? I included the VFS library and server in the same `src/vfs` package, I figure that the server is ubiquitous enough to include with the library. The server is also about half the size it used to be.

nfeske, 2018-04-23:
> I added a symbols file for the VFS library and API package, is this enough?

Yes, that's it.

I agree with including both VFS server and lib in the VFS source archive. Cool that the effect is already noticeable for the server's binary size.

As a last step, it might be worth revisiting the existing `used_apis` files - looking for occurrences of the combination `block_session`, `terminal_session`, `rtc_session`, `file_system_session`, which had to be included in src archives in order to use the VFS. Now, with your change, those entries could be deleted thereby loosening the coupling of those src archives from the individual session APIs. From the top of my head, examples are depot_query, menu_view, and terminal.

chelmuth, 2018-05-09:
@ehmry @ssumpf turning VFS into vfs.lib.so forces all VFS users to initialize potential static constructors by calling `exec_static_constructors()`. This is because of the check in ldso (https://github.com/genodelabs/genode/blob/staging/repos/base/src/lib/ldso/main.cc#L391) that returns true if there is a linker object beside ldso and the binary. My question is: Is this a hard dependency or could we find means to detect it there are really initializers to execute in the shared object (not just the hook function)?

chelmuth, 2018-05-09:
My test case is aeed399fe7ea7e8b525058a2e303a72fb2557e33 with the following commands

```
<genode>//tool/depot/create CROSS_DEV_PREFIX=genode-x86- UPDATE_VERSIONS=1 FORCE=1 -j8 genodelabs/bin/x86_64/base-linux genodelabs/pkg/x86_64/drivers_interactive-linux genodelabs/bin/x86_64/vfs
make run/terminal_echo KERNEL=linux
```

This shows the terminal works without calling any static constructors in vfs.lib.so despite it uses the `<fs>` plugin to access the VFS server.

ehmry, 2018-05-14:
Ping me when this is certain, and I will make the fixup patch for the VFS users.

commit 760937a3 (2018-05-03): Convert static VFS library to dynamic library

Fix #2759
commit e05602ae (2018-05-14): Convert static VFS library to dynamic library

Fix #2759
commit e2661c58 (2018-05-30): Convert static VFS library to dynamic library

Fix #2759
closed 2018-05-31 by chelmuth