genodelabs/genode #2760: static fs_rom

ehmry, 2018-04-12:
The Sculpt depot uses `fs_rom`, and with the new VFS notification patches for Rump, editing files in the depot can cause runtime changes in active subsystems, whereas these ROM sessions should be expected to be static. @nfeske and I had discussed offline an additional fs_rom that never updates ROMs, and caches and reuses dataspaces between clients. We can't share ROMs yet without a dataspace 'sealing' feature, but I think its worth to moving forward with an early implementation.

Caching ROM reads is already possible. Each session could store a hash of the ROM dataspace at the moment after the dataspace is populated and before the ROM capability is delegated to the client. When new sessions are created the current sessions would be searched for a session with the same label, the dataspace content of the existing session would be copied to the new session, and if the hash of the content in the new dataspace still matches the initial hash of the existing session, the dataspace is delegated, otherwise it is read again from the file-system.

Adding a hash function adds some complexity but should be justified by the performance gain from reduced file-system I/O. I would use BLAKE2 as the hash function, it hashes fast and is easy to port. https://blake2.net/

nfeske, 2018-04-12:
I'd skip the hashing but simply keep a registry of dataspaces that can be accessed by the ROM session's `label_last` string as key, and probably a reference count to free the memory of no-longer used ROMs.

Until we have the sealing of RAM dataspaces implemented, we must trust clients to never modify the content. The use of hashing would not change that. E.g., assume that two clients request the ROM for libc.lib.so. One client can modify the code within the library at any time and thereby manipulate the execution of the other client.

Even though we cannot not deploy the `static_fs_rom` in a serious scenario right now, I'd very much appreciate you implementing it. So we can already get a feeling for the conserved memory and the improved performance. The second part of the puzzle (the sealing of RAM dataspaces) can be implemented independently.

alex-ab, 2018-04-12:
> The second part of the puzzle (the sealing of RAM dataspaces) can be implemented independently.

We should finish #1633 - it already contains 2 (outdated) attempts.

ehmry, 2018-04-12:
I have new `static_fs_rom` component that is compatible with `run/depot_deploy` on my **issue2760_static_fs_rom** branch. `static_fs_rom` is trivial, but the patch preceding is not, de6dd853b9fa8dcdc6ab335db40f0f226f82b9e8 is a utility that I made earlier for processing session requests. It could be part of a new session management API or I could make it private to `static_fs_rom` for now.

ehmry, 2018-04-12:
To clarify my comment on hashing ROMs, hashes would only be made of dataspaces that have not been delegated to clients, and no clients would share the same dataspaces. I would copy the ROM content to a new private dataspace and then check that for changes.

nfeske, 2018-04-12:
I like your initiative regarding the session-request utility but would prefer to keep it private to the component until I have tried it out for some existing components of different flavors. In particular, we should consider scenarios where the session response cannot immediately be delivered but at a later point (e.g., when the session creation depends on an external event). So in general, more than one session creation may be in flight. The current version seems to work synchronously, doesn't it?

ehmry, 2018-04-12:
The current version of my session-request utility does no implicit session delivery, the documentation I made was outdated. The current version of this fs_rom will not deliver a session capability until the ROM dataspace is populated, and can populate dataspaces concurrently (not quite tested though).

I've updated my branch to make the session-requests head component-local.

nfeske, 2018-04-12:
Ah, now I got it. I completely missed the `deliver_session_cap` call in the `Session_component`. That is pretty cool!

Just to double check that I understood the hashing correctly. Each client gets a different RAM dataspace. But instead of fetching the content from the file system each time, the content is copied from one of the existing dataspaces if available. By comparing the hash after copying with the original content's hash, the server validates that the first client has not messed with the dataspace in the meanwhile. Hence, the `static_fs_rom` does not preserve RAM but at least saves the time for the file-system operations. If this is how it works, I like this idea as a nice middleground until we have the sealing covered. With the sealing, neither the hashing nor the copying would be needed anymore. @ehmry can you confirm that I got this correctly?

ehmry, 2018-04-12:
Yes, that is correct. I don't suspect that there would be problems with sharing dataspaces but I would rather spend the extra memory to be safe.

ehmry, 2018-04-12:
Feels a little bit faster ;-) Depot recipes on my `sculpt_next` branch.

ehmry, 2018-05-16:
I have a new version of `static_fs_rom` using read-only region-map attachments at 198b5f302d0b79130c1ebc4205d6510df41caf2c on my **issue2760** branch. I have not yet been able to get Sculpt next running well enough to test it.

ehmry, 2018-05-24:
@alex-ab I have adjusted `run/depot_deploy` to use `static_fs_rom`, but get overmap errors when using NOVA, see 0eea5f6441e196f9106e49fde7bbd869913a3c10.

alex-ab, 2018-05-24:
@ehmry: i'll take care

ehmry, 2018-06-11:
@alex-ab I tested static_fs_rom with sculpt and got some interesting errors. After loading a few ROMs I get a warning from NOVA, and then ram_fs faults, followed by the other components that I could interact with. What is interesting is that they all fault at the same stack address, and all seem to fault from within the base library:

```
[ 0] Killed EC:0xffffffff8174b700 SC:0xffffffff81901d60 V:0xa0 CS:0x2b IP:0x4c20e(0x80b32) CR2:0xffffffff8174b808 ERR:0x0 (Livelock)
page fault, pd='init -> runtime -> ram_fs' thread='root' cpu=0 ip=0xbaed3 address=0xa03fafe8 stack pointer=0xa02fef18 qualifiers=0x6 irUWp reason=1
page fault, pd='init -> runtime -> depot_rom' thread='ep' cpu=0 ip=0x4b45a address=0xa01edff8 stack pointer=0xa02fef18 qualifiers=0x6 irUWp reason=1
page fault, pd='init -> runtime -> storage browser noux' thread='ep' cpu=0 ip=0xba81a address=0xa01edfe8 stack pointer=0xa02fef18 qualifiers=0x6 irUWp reason=1
page fault, pd='init -> runtime -> nic_router' thread='ep' cpu=0 ip=0xb9f6a address=0xa01edfe8 stack pointer=0xa02fef18 qualifiers=0x6 irUWp reason=1
```

Running the test again gives me similar errors, but with a different stack pointer:

```
[ 0] Killed EC:0xffffffff81745700 SC:0xffffffff81b16580 V:0x9e CS:0x2b IP:0xc4a44(0x80b32) CR2:0xffffffff81745808 ERR:0x0 (Livelock)
no RM attachment (READ pf_addr=0x5 pf_ip=0x5 from pager_object: pd='init -> runtime -> wm' thread='wm')
page fault, pd='init -> runtime -> wm' thread='wm' cpu=0 ip=0x5 address=0x5 stack pointer=0x0 qualifiers=0x14 IrUwp reason=1
page fault, pd='init -> runtime -> depot_rom' thread='ep' cpu=0 ip=0x4b45a address=0xa01edff8 stack pointer=0x0 qualifiers=0x6 irUWp reason=1
page fault, pd='init -> runtime -> nic_router' thread='root' cpu=0 ip=0xbaed3 address=0xa03fafe8 stack pointer=0x0 qualifiers=0x6 irUWp reason=1
```

alex-ab, 2018-06-11:
We encountered this effect also for the fs_rom, and avoided it by ff4fa8acd13877f3c5bca16f4c7c0db04abf0b75. The issue is, that during the synchronous RPC calls (like dataspace()) the static_fs_rom must not call the parent - otherwise you get a IPC call&help loop which the kernel (correctly) detects as livelock. 
So, the static_fs_rom server have to open sessions (like LOG) solely during the startup of static_fs-rom or in the constructor of a session.

ehmry, 2018-06-11:
Funny, I looked at that commit, but I decided to test against master. This is good to know.

ehmry, 2018-07-03:
See 0650dd883a104558cdf5aca781c1e31db95d6713.

nfeske, 2018-07-03:
Thank you. Merged to staging right away.

nfeske, 2018-07-04:
The `cached_fs_rom` fails to build on 32-bit:

```
repos/os/src/server/cached_fs_rom/main.cc:89:59: error: narrowing conversion of ‘((Cached_fs_rom::Rom_file*)this)->Cached_fs_rom::Rom_file::_file_size’ from ‘File_system::file_size_t {aka long long unsigned int}’ to ‘Genode::size_t {aka long unsigned int}’ inside { } [-Werror=narrowing]
   Attached_ram_dataspace _file_ds { _env.pd(), _env.rm(), _file_size };
                                                           ^~~~~~~~~~
```

nfeske, 2018-07-04:
Just added a fixup to staging that fixes the build on 32 bit.

ehmry, 2018-07-13:
I have a critical patch for a number of bugs at 
4da4b4a8f8f28f734de040305ddbdc6806c8e7b3.

ehmry, 2018-07-13:
Er, 8e2ba309323d80b9442497b5f1dd9a37a8f281fc is better as it does not touch the sculpt_manager.

commit 96e55914 (2018-07-03): Cached_fs_rom: serve static ROM sessions from a cache

This component is contrasted with the fs_rom server that serves
independent dataspaces to each client. Using a cache was not possible
until the region map session supported the creation of read-only
attachments.

Test at run/read_only_rom.

Ref #1633
Fix #2760
commit 09bf68e8 (2018-08-02): Cached_fs_rom: serve static ROM sessions from a cache

This component is contrasted with the fs_rom server that serves
independent dataspaces to each client. Using a cache was not possible
until the region map session supported the creation of read-only
attachments.

Test at run/read_only_rom.

Ref #1633
Fix #2760
closed 2018-08-08 by nfeske