genodelabs/genode #2761: Merge `Input::Event::CHARACTER` into `Input::Event::PRESS` events

nfeske, 2018-04-12:
In #2264, I added the `CHARACTER` event type for delivering symbolic character events to input clients. In contrast to low-level PRESS/RELEASE events, which correspond to physical buttons, CHARACTER events take the keyboard layout into account. Such events are typically supplemented into an input-event stream by the input-filter component.

In the meanwhile, we gathered practical experience with this approach. The findings are:

* The approach of supplementing the input stream with symbolic information by the input-filter works as intended. It gives us a clean way to configure keyboard layouts and key repeat at a central place.
* The CHARACTER event type greatly simplifies the code of components that consume raw character streams such as the terminal.
* In Qt, both CHARACTER and low-level PRESS/RELEASE events are processed but usually serve different purposes. E.g., characters are used when typing input into a text-entry field (thereby applying the keyboard-layout configuration of the input filter) whereas cursor keys are usually handled via PRESS/RELEASE (e.g., to control a game). Unfortunately, the distinction between both cases is not always clear-cut. For example, when running JavaScript in WebKit (e.g., opening google.com in Arora), both the low-level PRESS/RELEASE events as well as the CHARACTER events are processed. The former type of events are actually translated to characters by Qt's built-in keyboard-layout handling (which is not our intention). As a result, characters happen to appear twice when typing text in the search field.
* In libSDL, a press event contains both the hardware scancode of the pressed key as well as the symbolic information associated with this key. With the distinction of  PRESS from CHARACTER events, only one part of this information is immediately available when an event occurs. This causes some applications to misbehave. E.g., the atari800 emulator won't accept any input that requires the shift key because such input is not handled by tracking the physical shift-key state but rather by using the symbolic information of the actual character (like '"'). In principle, this problem can be fixed by enhancing the libSDL backend with a state machine that expects pairs of CHARACTER and PRESS events. But this built-in heuristics defeats my original intention to keep the input-interface simple.

Based on these experiences, I came to the conclusion that it is better to incorporate symbolic information as an optional attribute into PRESS events.

* Each PRESS event potentially carries an Unicode codepoint whenever a symbolic interpretation is known.
* PRESS events for the `KEY_UNKNOWN` can still carry a codepoint. This is useful for character repeat events. For each PRESS of a `KEY_UNKNOWN`, there should also be a RELEASE event to satisfy the client's expectations that the number of press events equals the number of release events over time.

ehmry, 2018-04-12:
This also caused problems with TyrQuake input settings (they've been fixed though).

nfeske, 2018-04-22:
I merged ff70e68db7f91ed54d0d3eb80b94caaaab617dfc to staging. The patch replaces the `Input::Event` by a completely new interface, and it contains adaptations of all input-processing components I am aware of.

Since the event API is completely new, some of those adaptations (in particular the back ends of Qt or VirtualBox) are almost rewrites, which comes at the risk of possible regressions. I am particularly unsure of the touch handling since I could not test the new code.

@alex-ab May you give the multi-touch handling of Qt and VirtualBox a spin to see if I introduced any damage here?

For Qt, I opted for removing the dependency from the Qt's existing evdev keyboard handler, which removes a lot of uncertainties (especially wrt to key-repeat handling of non-symbol keys). The only thing that I deliberately left out for now is the modifier-state tracking (in particular shift and control). E.g., holding shift while pressing the cursor keys in qt_textedit does not select any text. @cproc Could you try supplement the modifier-state tracking? I think this could be done analogously to how the mouse-button state is already tracked now.

cproc, 2018-04-23:
Commit 5ff4890 implements the modifier-state tracking for Qt.

ehmry, 2018-04-24:
Fixup for `menu_view` at c877a6524b339564549fb9bd403adac22a9474b3.

alex-ab, 2018-04-26:
@nfeske: vbox5 and multitouch seems fine - qt5 has not multitouch support, as far as i know

alex-ab, 2018-04-26:
@nfeske: at least the qt5/Genode version

nfeske, 2018-04-26:
@alex-ab Thank you for confirming that I did not break anything fundamental. Good to hear! ;-)

Qt's nitpicker backend actually has some "touchy" stuff in it, which I also reworked. Maybe @cproc knows best how far Qt's touch support is supposed to go?

nfeske, 2018-07-02:
Fixed in version 18.05

closed 2018-07-02 by nfeske