genodelabs/genode #2763: Session-request handler API

nfeske, 2018-04-13:
As part of issue #2760, @ehmry revived the API modernization effort regarding the last remaining visible legacy of the old API, which is the server-side root interface.

Since we changed the parent interface to be asynchronous, the synchronous root interface does not actually exist anymore but it is merely emulated using a component-local proxy mechanism. It goes without saying that this proxy mechanism was just meant as a bridging solution until we have modernized this part of the API as well.

@ehmry's proposal is quite nice, especially because it supports the out-of-order creation of sessions. However, I am hesitant to adopt it as is for the official API for the following reasons:

* The old session-args form of passing session-construction arguments should eventually be replaced by more robust form (issue #2689 supports this observation). But this is a longer-term effort. With the change of the asynchronous parent interface, I already moved the handling of the session label and quotas elsewhere. But regular session arguments like packet-buffer sizes are still communicated via in the `Arg_string` format. In the future, I would like to present such information in the form of an XML node to the server and get rid of `util/arg_string.h`. To anticipate this direction, it might be sensible to actually pass the `<args>` node to the handler.

* The way of how the `Session_requests_handler` is invoked differs from Genode's existing patterns, which makes it a bit unintuitive. The approach of mixing-in protected functions into the `Main` class is a little too much magic in my opinion. I would prefer to split the mechanism into two dead-simple parts instead. The first part is a pure-abstract `Session_request_handler` (note the singular) interface with the virtual methods `handle_session_create`, `handle_session_upgrade`, `handle_session_close`. This way, `Main` could implement this interface without the need to use any lambdas. The second part is the actual mechanism that opens and watches the ROM and calls the `handle_session_*` functions. This could be instantiated as a member variable of main. The second part could be called `Session_requests_rom` (corresponding to the underlying mechanism) but maybe there is a better name for it?

* I want to gradually change the server implementations to use the `Session_object` class (see https://github.com/genodelabs/genode/blob/master/repos/base/include/base/session_object.h) instead of `Rpc_object`. Doing this in concert with the switch to the `Session_request_handler` make a lot of sense. Right now, `Session_object` is mostly used within core but it is meant to simplify all servers, especially regarding the session-quota handling. Given this background, I find it important that the `Session_request_handler` fits perfectly together with this class. I.e., the constructor arguments (resources, label, diag) should be readily provided to the `handle_session_request` operation to be passed to the session-component object (and thereby to the `Session_object`).

* Even though the session-request handler supports the out-of-order creation of sessions, the common case will definitely be the synchronous way. It would be nice to alleviate the need to manually call of `parent.deliver_session_cap` in this case. But I am not completely sure about how to achieve that. Maybe it even makes sense to make the session-request handler aware of the state of the `Session_object` wrt whether the session cap is ready to be delivered or not yet?

That all said, I am not in a hurry (and merely use this issue to explain myself). Before we promote a new API, all of the points above should be considered first. I don't want to risk iterating publicly over these all-encompassing parts of the API multiple times.

ehmry, 2018-04-13:
Now that you explain it I would favor using a pure-abstract Session_request_handler object rather than lambdas. My intermediate utility was a conversion from Nim to C++, Nim not supporting virtual methods, I was using a Nim iterator.
https://github.com/genodelabs/genode-world/blob/e6c31ccc3ec41de8d5b86228e656e8081874578d/include/nim/genodeservers.nim#L155

The other benefit of moving away from the Root interface is the component gets unique session IDs for free, and that makes it easy to keep sessions in ID spaces.

nfeske, 2018-06-21:
The observation of issue #2877 is relevant for this discussion. Session-close requests must always be processed before session-create requests.