genodelabs/genode #2767: SHA1 deprecation

ehmry, 2018-04-16:
https://shattered.io/

It is known to be practical to forge SHA1 checksums on files without a bruteforce attack, so it would be wise to move to a different hash function for verify ports in the near future.

I would volunteer to add support for an additional verification method in the `prepare_ports` tool. I think SHA-256 would be the least disruptive alternative.

nfeske, 2018-04-16:
Thanks for bringing up this issue and for volunteering. SHA-256 sounds good to me.

ehmry, 2018-04-18:
Patch at c03ab9d8a95a693568c67aca6df03f2b2c32bd09. This adds support for 'SHA256' in port files and leaves the port and depot versioning intact.

ehmry, 2018-04-19:
Yes, new patch at 3b3fae6dd7ea68aa63c806cd22db0d448b4c006d to keep the `SHA` variable name, yet still refusing to check SHA1 during `CHECK_HASH=no`.

nfeske, 2018-04-19:
Thanks for the new commit, that's very good!

ehmry, 2018-05-07:
I have two more commits for this series, b85b00014f4cb60dd4f58a73b40302aa5f9265d5 replaces SHA1 file digests with SHA256 and 2f4a8bf55831dc9ef598c06afe4d4b821866201c removes SHA1 verification.

ehmry, 2018-06-04:
Fixed in 18.05.

closed 2018-06-04 by ehmry