genodelabs/genode #2777: Timer::Connection: option for using non-I/O signals

nfeske, 2018-04-18:
Currently, the timeout API provided by the `Timer::Connection` operates on the `Signal_context::Io` level, not the `App` level. This is desired in situations where timeouts are a part of an low-level I/O state machine like inside the libc. But when used by an application, timer handlers should better be executed at the `App` level to avoid interference with, i.e., the VFS.

In my case, I hit this issue while changing the terminal to use the VFS. Under certain conditions (resizing the terminal, which triggers a re-construction of the font handling), the so-called "flush" handler was triggered while the VFS was blocking as part of the re-construction. This is not expected. The flush handler, in turn, accessed the - now incompletely constructed - font handling.

I propose adding an optional `Signal_context::Level` argument to the `Timer::Connection` constructor that would allow us to explicitly specify the desired signal-handling level. In my opinion, it would be sensible to use the `App` level by default.

jschlatow, 2023-12-06:
I also hit this issue when calling `clock_gettime()` from within a timeout handler. I have the feeling this could become a quite irritating pitfall for framework users. I'd therefore also favour using `App`-level signal handling by default.

nfeske, 2023-12-07:
Thank you @jschlatow for breathing new life into this issue!

Maybe we can drop the I/O signal variant altogether? In the few cases where a timer session is used for I/O signals, one is still free to manually install an `Io_signal_handler` w/o using the `One_shot_timeout` and `Periodic_timeout` utilities.

When changing this, however, we need to revisit the places where these utilities are used today. While grep'ing for them, only a few places look like they could depend on the I/O signal level, specifically code that is executed as part of VFS plugins. When hosting such plugins in a POSIX application, no application-level signal handler is ever executed. E.g., the `dde_rump/src/include/rump/timed_semaphore.h` would never trigger in this case. Since those cases are rare enough, we could adapt them to the manually installed `Io_signal_handler`.

The background of my tendency to not offer I/O-level timer handlers is that the risk of executing global side effects from those handlers is too high. It is a bug prone pattern that is hard to spot. In practice, we often elevate the signal level to the application-level (by calling local_submit) anyway. I ultimately came to the conclusion that the ease of use of  `Io_signal_handler` was probably a mistake. Consequently, I think that the `One_shot_timeout` and `Periodic_timeout` should only operate at the application-signal level.

@m-stein and @jschlatow what do you think?

chelmuth, 2023-12-08:
I was not sure about the  effects on dde_linux, but with the current schedule/execute scheme this change seems to be fine for this case too. @skalk @cnuke do you agree?

nfeske, 2023-12-08:
> Is this what you are referring to with the global side effects [...]?

Yes.

cnuke, 2023-12-11:
I am going to take a look if we can also get rid of the `wait_and_dispatch_io_signal` or at least boil it down to situation where we have to wait for an IO-signal (e.g. an interrupt) in the GPU drivers.

cnuke, 2023-12-12:
AFAICT, backed by testing with available presets, `wait_and_dispatch_io_signal` is no longer necessary in the _lima_gpu_drv_, I'll check the _etnaviv_gpu_drv_ next.

cnuke, 2023-12-12:
The way the _imx8mq_gpu_drv_ is build currently we still rely on the handling of I/O signals via `wait_and_dispatch_io_signal` in case of interrupts but not timeouts. So we could remove I/O signal based timeouts within the Lxkit.

nfeske, 2026-05-08:
This is along-standing issue which has been neglected far too long. The `Periodic_timeout` and `One_shot_timeout` utilities still operate on I/O level, which is risky (and may be unintended).

Let me propose to address it as follows:

1. Rename the parts of the timeout API to better reflect the reality
   `One_shot_timeout` -> `One_shot_io_timeout`
   `Periodic_timeout` -> `Periodic_io_timeout`
   `Timeout_handler` -> `Io_timeout_handler`

2. Introduce a new `One_shot_timeout` class that aggregates an `One_shot_io_timeout` and an (application-level) `Signal_handler _timeout_handler`). In its private `_handle_io_timeout` method, it remembers the current-time argument (in a private member) and issues a `_timeout_handler.local_submit()`. This lifts the execution from I/O level to application level. The (application-level) private `_handle_timeout` method then calls the user-provided handler function with the remembered current time as argument.

3. Grep the code base for the use of `*_io_timeout` and reconsider. All places that naturally expect application-level execution should be changed to the use of the non-I/O `timeout` utility.

@jschlatow, may you be interested in pursuing this plan?