genodelabs/genode #2782: vfs_cow crashes vfs server when COW dir is in VFS root

senier, 2018-04-19:
When the `vfs_cow` plugin is configured such that the COW dir (the merged one) is in VFS root and the mutable and the immutable directories are subdirs of that very VFS root, the VFS server crashes:

```xml
            <vfs>
                <dir name="immutable">
                    <rom name="timer"/>
                </dir>
                <dir name="mutable">
                    <ram/>
                </dir>
                <!-- <dir name="cow"> -->
                    <cow ro="/immutable" rw="/mutable"/>
                <!-- </dir> -->
            </vfs>
```

When uncommenting the commented out subdir, the issue vanishes. Not sure whether this configuration should be allowed. The crash is as follows (Qemu, Nova, x86_64):

```
page fault, pd='init -> vfs' thread='ep' cpu=0 ip=0x107f09b address=0xa01ecab8 stack pointer=0xa04fef18 qualifiers=0x6 irUWp reason=1
```

[Here](https://github.com/Componolit/componolit/blob/android_runtime/run/mfe/cow_root.run) is a reproducer.

ehmry, 2018-04-20:
Yes, the COW plugin is not very well tested and doesn't contain any loop detection. I will take a look.

ehmry, 2018-04-23:
Fixed on my `issue2782_cow`. I'd like to sneak in  28b47f4a2b3cfc4d98e608169b99410fba9abbba as well to fix an issue with the `Dir_file_system` plugin potentially closing handles at the wrong file-systems, which was detected by the COW plugin.

nfeske, 2018-04-27:
I reverted the commits 6557e71f2cc31391cac2863d25274f3a8c4baab6 and 02fdf076493679f5e258ece757c5ff89c19aacf7 from staging.

The former causes problems with the depot_deploy.run script. Sooner or later we'll need to address the dynamic VFS configuration as a well planned step. Until then, let us keep the configuration at construction time except for a few exceptions like socket-fs config changes. However, I kept the root-dir-detection part of the patch in the form of 7eabe27ada211c4058fab9f7495548944ac38a49 since I very much welcome with this part. ;-) @ehmry are you fine with that?

Regarding the latter commit, it solves the problem at hand, but not in a way I am confident with. I think we should document the current limitation of the VFS instead: A plugin may refer to other parts of the VFS but only to those that appear _before_ its own configuration node. By that rule, circular dependencies are effectively ruled out. Of course, it would be nice to detect the violation of this rule at runtime, but this would be merely a convenience feature, not a must-have-for-now in my opinion.

ehmry, 2018-04-30:
@nfeske yes, I'm fine with waiting for a more elegant solution. Checking plugin ordering is easy to implement however, patch at 0a8fa0d96db641e9bd0b56b3bbe1cbc911f11a03.

nfeske, 2018-08-24:
We replaced the cow plugin with the import plugin in the meanwhile.

commit 02fdf076 (2018-04-25): VFS: loop detection for COW plugin

Use the initial values of a stat buffer as a side-channel for the plugin
to communicate with itself. Detection only occurs during 'apply_config'.

Fix #2782
closed 2018-08-24 by nfeske