genodelabs/genode #2783: Fix rump fd leaks and memory pressure

cnuke, 2018-04-20:


cnuke, 2018-04-20:
@nfeske please try above commits.

chelmuth, 2018-04-24:
@cnuke please have a look at nightly errors like (qemu/hw_x86_64)
```
[2018-04-24 05:11:56] [init -> rump_fs] resource_request: ram_quota=2101248
[2018-04-24 05:11:56] [init] child "rump_fs" requests resources: ram_quota=2101248
[2018-04-24 05:11:56] [init -> rump_fs] resource_request: ram_quota=2101248
[2018-04-24 05:11:56] [init] child "rump_fs" requests resources: ram_quota=2101248
[2018-04-24 05:11:56] [init -> rump_fs] Warning: backend allocator exhausted (out of RAM)
```

cnuke, 2018-04-24:
@chelmuth please merge 1a635d5 to staging.

chelmuth, 2018-04-25:
@cnuke please look into the following on x86_32
```
server/rump_fs/file_system.cc:93: undefined reference to `rump_set_memlimit(unsigned long)'
```

cnuke, 2018-04-25:
@chelmuth please merge 3715374 to staging.

commit 97436442 (2018-04-20): rump: add knob to set memlimit

By now, rump would query its available RAM quota to determine the
memory limit minus some RAM reserved for Genode meta-data. This
does not work when the VFS rump plugin is used as the available
quota belongs to the VFS server. In this case the memlimit should
be set by specifing the RAM in the plugin's config, e.g.:

! <vfs>
!   <rump fs="ext2fs" ram="64M" writeabl="yes"/>
! </vfs>

Fixes #2783.
closed 2018-05-03 by chelmuth