genodelabs/genode #2790: Libc SIGHUP triggers

ehmry, 2018-04-26:
There is a fair amount of existing 'server' software that would be nice to run natively with minimal modification. DNS resolvers are an example. It would be convenient to package this software for the Depot in a way that puts some hard restrictions of the components, while leaving the nitty-gritty application details to be configured by the user at runtime.

Retaining existing configuration files and formats for ported software is practical, but we have no means to trigger configuration updates without restarting such a server. The unofficial standard for unix software is to reload or restart using the *SIGHUP* signal. Perhap the libc library could support *SIGUP* by configuring a list of files to watch at the VFS layer that would trigger a signal.

Example of a component with a updatable default configuration file:
```
<config>
	<libc>
		<signal name="SIGHUP" watch="/etc/foo.config"/>
	</libc>
	<vfs>
		<dir name="etc-default"> <tar name="etc.tar"/> </dir>
		<dir name="etc-fs"> <fs label="etc"/> </dir>
		<dir name="etc"> <cow ro="/etc-default" rw="/etc-fs"/> </dir>
	<vfs>
</config>
```

Adding special support for one signal type doesn't feel good, but right off I can't think of other signals worth bothering with. The first step would be of course to add signal support to the libc.

nfeske, 2018-04-26:
That is a very elegant idea - really cool. It brings both worlds - POSIX and Genode - almost seamlessly together.

Solid POSIX signaling support is also very much anticipated, I know for sure.

chelmuth, 2019-12-20:
The libc now supports signal handling, which removes the major blocker of this feature.

nfeske, 2020-05-08:
The issue is two years old and has become stale. Even though, the idea is nice, I'm closing the issue because there is no plan to work on it. We can still resurrect it once we have an actual need for it.

closed 2020-05-08 by nfeske