genodelabs/genode #2799: depot: optional archive version for publish tool

rite, 2018-05-02:
In order to simplify the automation of publishing archives, I would like to use tool/depot/publish tool without specifing the archive version.

nfeske, 2018-05-02:
The publish tool strictly does not touch anything within repos/ but works on the depot/ and public/ directories only. I'd like to keep this clear cut between the extraction of archives from the source tree and the publish/download steps.

In my opinion, the proper way to address your request would be a new `tool/depot/publish_current` utility that uses the existing `publish` tool internally while obtaining the version info automatically.

rite, 2018-05-02:
I abandoned the 'version' tool and added 'publish_current' as suggested by @nfeske. I don't have much experience writing make scripts and providing both tools would just result in code duplication... So if a standalone 'current'/'version' tool is desired, I leave it up to others...

commit 058b92ca (2018-05-31): depot: tool for publishing current versions

In order to simplify the automation of publishing archives, this tool
publishes the current version as specified in the recipes.

Fixes #2799
closed 2018-05-31 by chelmuth