genodelabs/genode #2801: platform_drv does not rewrite IRQs in VirtualBox VM with ICH9 chipset

cproc, 2018-05-02:
The ACPI report says ```bridge_bdf="0x0"``` in each routing entry, but there is no PCI bridge at 00:00.0 and the following check fails: https://github.com/genodelabs/genode/blob/47569458d444999225b2f6541e03606160200d6f/repos/os/src/drivers/platform/spec/x86/pci_session_component.h#L1000-L1006

'lspci' in a Linux VM shows for ICH9:

```
00:02.0 VGA compatible controller: InnoTek Systemberatung GmbH VirtualBox Graphics Adapter
00:03.0 Ethernet controller: Intel Corporation 82540EM Gigabit Ethernet Controller (rev 02)
00:04.0 System peripheral: InnoTek Systemberatung GmbH VirtualBox Guest Service
00:05.0 Multimedia audio controller: Intel Corporation 82801AA AC'97 Audio Controller (rev 01)
00:07.0 Bridge: Intel Corporation 82371AB/EB/MB PIIX4 ACPI (rev 08)
00:18.0 PCI bridge: Intel Corporation 82801 Mobile PCI Bridge (rev f2)
00:19.0 PCI bridge: Intel Corporation 82801 Mobile PCI Bridge (rev f2)
00:1f.0 ISA bridge: Intel Corporation 82801GBM (ICH7-M) LPC Interface Bridge (rev 02)
00:1f.1 IDE interface: Intel Corporation 82371AB/EB/MB PIIX4 IDE (rev 01)
00:1f.2 SATA controller: Intel Corporation 82801HM/HEM (ICH8M/ICH8M-E) SATA Controller [AHCI mode] (rev 02)
00:1f.4 USB controller: Apple Inc. KeyLargo/Intrepid USB
00:1f.5 USB controller: Intel Corporation 82801FB/FBM/FR/FW/FRW (ICH6 Family) USB2 EHCI Controller
```

and for PIIX3:

```
00:00.0 Host bridge: Intel Corporation 440FX - 82441FX PMC [Natoma] (rev 02)
00:01.0 ISA bridge: Intel Corporation 82371SB PIIX3 ISA [Natoma/Triton II]
00:01.1 IDE interface: Intel Corporation 82371AB/EB/MB PIIX4 IDE (rev 01)
00:02.0 VGA compatible controller: InnoTek Systemberatung GmbH VirtualBox Graphics Adapter
00:03.0 Ethernet controller: Intel Corporation 82540EM Gigabit Ethernet Controller (rev 02)
00:04.0 System peripheral: InnoTek Systemberatung GmbH VirtualBox Guest Service
00:05.0 Multimedia audio controller: Intel Corporation 82801AA AC'97 Audio Controller (rev 01)
00:06.0 USB controller: Apple Inc. KeyLargo/Intrepid USB
00:07.0 Bridge: Intel Corporation 82371AB/EB/MB PIIX4 ACPI (rev 08)
00:0b.0 USB controller: Intel Corporation 82801FB/FBM/FR/FW/FRW (ICH6 Family) USB2 EHCI Controller
00:0d.0 SATA controller: Intel Corporation 82801HM/HEM (ICH8M/ICH8M-E) SATA Controller [AHCI mode] (rev 02)
```

But the PIIX3 chipset cannot be used for Genode VMs anymore because it does not support memory mapped PCI configuration space access.

cproc, 2018-05-02:
'vbox5_genode_usb_hid.run' (issue #2802) can be used as a test.

chelmuth, 2018-05-02:
The ACPI specification defines *Source* and *Source Index* like follows in *6.2.13 _PRT (PCI Routing Table)*

> Source: If this field is the integer constant Zero (or a Byte value of 0), then the interrupt is allocated from the global interrupt pool.
> Source Index: If the Source field is the Byte value zero, then this field is the global system interrupt number to which the pin is connected.

So, the routing entry is not refering BDF 0:0:0 but the GSI directly. The PCI spec does not require BDF 0:0:0 to be existent.

alex-ab, 2018-05-04:
With the commit it is fixed for me - tested on some Intel and one AMD machine.  Genode/Virtualbox5 (ICH9 chipset)  works now with Genode guests, so network and sata driver is now working for me.

alex-ab, 2018-05-07:
@chelmuth: thx for reviewing, I adjusted the commit

commit de43a83f (2018-05-07): x86: detect root bridge properly

Beforehand the root bridge was expected to be at 0:0.0.

Fixes #2801
closed 2018-05-31 by chelmuth