genodelabs/genode #2807: Libc: 'sendto' with address fails for TCP

ehmry, 2018-05-03:
Stubby seems to use common code for sending requests through UDP and TCP sockets, because it is calling `sendto` on TCP sockets with a destination address argument. The libc returns an error because the destination address of a TCP connection is immutable, but I see no reason not to simply ignore the address argument in this case (and blame the application if something bad happens).

ehmry, 2018-05-03:
Fixed at ebc3b3ff073c886903465daabba49095b7db9ff6. This enables Stubby to make requests over TCP and TLS.

chelmuth, 2018-05-07:
I merged the commit as I agree with your rationale. Nevertheless the standard seems to disagree (from http://pubs.opengroup.org/onlinepubs/9699919799/)

> If the socket is a connectionless-mode socket, the message shall be sent to the address specified by dest_addr if no pre-specified peer address has been set. If a peer address has been pre-specified, either the message shall be sent to the address specified by dest_addr (overriding the pre-specified peer address), or the function shall return -1 and set errno to [EISCONN].

Opinions? What would happen in stubby on `EISCONN`?

chelmuth, 2018-05-07:
The linux man page is less strict which backs your idea.

> EISCONN
              The connection-mode socket was connected already but a recipient  was  speci‐
              fied.   (Now either this error is returned, or the recipient specification is
              ignored.)

ehmry, 2018-05-08:
I think the standard implies that the address argument has no effect for TCP, but an error would be returned for UDP if `connect` has been used.
> If the socket is connection-mode, dest_addr shall be ignored.

What Stubby is in fact trying to do here is skip `connect` and use `sendto` with a [TCP Fast Open](https://en.wikipedia.org/wiki/TCP_Fast_Open) optimization that is not defined in the standard. There seems to simply be a faulty `ifdef` path in the libgetdns code that uncommon on recent POSIX.

I would leave the patch to ignore the TCP address, send a patch to libgetdns, and make sure that `EISCONN` is returned for UDP sockets that are in a 'connected' state.

ehmry, 2018-05-08:
Maybe we should just leave UDP `sendto` as it is, the spec seems to say `EISCONN` is optional.
> If a peer address has been pre-specified, either the message shall be sent to the address specified by dest_addr (overriding the pre-specified peer address), or the function shall return -1 and set errno to [EISCONN].

chelmuth, 2018-05-08:
Please help me I'm a bit confused: Do we stay with 3239bd059c61a066567e2c4cc19cd3d6e9477060 as it is or not?

ehmry, 2018-05-08:
Yes, I think 3239bd059c61a066567e2c4cc19cd3d6e9477060 is fine.

closed 2018-05-31 by chelmuth