genodelabs/genode #2811: Seoul unable to Multiboot non-ELF images

ehmry, 2018-05-07:
Seoul can Multiboot images in ELF format, but not support images in other binary formats, like 9front, which uses a non-gcc toolchain. 9front is a fringe case, but the [multiboot spec](https://www.gnu.org/software/grub/manual/multiboot/html_node/Header-magic-fields.html) requires that other formats be supported:
> Compliant boot loaders must be able to load images that either are in ELF format or contain the load address information embedded in the Multiboot header; they may also directly support other executable  formats, such as particular a.out variants, but are not required to.

Adding support for load address information isn't complicated, it just needs to be done right.
https://github.com/alex-ab/seoul/blob/ac983d9104792c5ba59106eaff242753b989506f/executor/vbios_multiboot.cc#L112

I will do what I can when I get the time, but if someone else beats me to it, I would be happy.

nfeske, 2020-04-23:
Is this feature something that anyone still wants? If not, I'd like to close this issue.

alex-ab, 2020-04-23:
@ehmry: I think this is specific to Seoul in the first place and not so much related to Genode. Feel invited to open up the issue again at https://github.com/alex-ab/seoul/issues in case you are still interested.

closed 2020-04-23 by alex-ab