genodelabs/genode #2815: nic_router: handle uplink link_state

BorisMulder-CSL, 2018-05-07:
We are experiencing the following problem:
We have a nic_router with some clients and an uplink connected to a nic_drv.
Whenever we plug out the network cable, the network packets are still sent from the router to the driver until the buffer is full. 
When we plug in the cable again, a few packets buffered by the nic_drv are sent over the wire directly, but any further packets are not sent and the clients are unable to send or receive any messages over the network from then on. Only restarting the system or restarting the nic_router seems to fix this.

Looking into the code of nic_router, on the uplink side it does not seem to do anything with the link_state and the link_state_sigh. But we think it should do something at least, like sending an ICMP 'destination unreachable' packets to the client if it tries to send packets to the uplink, perhaps propagating the link_state signal to the client side, or take some other action. At least the uplink domain should do the DHCP handshake again, as we might have plugged the computer into a different network. We think it should not send old buffered messages over the wire when it gets plugged in again, considering a user might send sensitive data over a public network after it was plugged out from a private network.

Another additional nice feature would be to include the link_state in the nic_router report. This way, some other components can know the current network state and act accordingly.

chelmuth, 2018-05-07:
After I experienced a similar scenario last Friday we revived the discussion about this topic internally. We will come up with a proposal after it settled.

m-stein, 2018-05-22:
This is a first way mark for solving this issue:
6ef8ae384 nic_router: when freeing a link log it beforehand
4773b8ae7 sculpt: fix debian URL in download runtime
cc6b474d3 net: get rid of unused mac_from_string method
3e63d49c8 config.xsd: adapt Boolean type to ascii_to(bool)
c780227cf virtualbox_nic_router.run: test uplink reconnect
f6c3a7526 autopilot.list: add nic_router_uplinks
865bd208b nic_router_uplinks.run: test uplink re-connect
5e2ef91a2 nic_router: re-use dynamic IPv4 config if possible
cce3c22da nic_router: label attribute for uplink domain
dacd47f28 nic_router: Ipv4_config::print method

The commits from "nic_router: Ipv4_config::print method" to "autopilot.list: add nic_router_uplinks" make it possible to re-connect the uplink with different back-ends and different IPv4 subnets or to completely remove it temporarily. It does not yet implement that the NIC router considers the link state of the uplink. The rest of the commits are some small things I stumbled over in the meanwhile.

m-stein, 2018-05-22:
I broadened the concern of this issue to "deal with state changes of the uplink in general". I hope this is fine?

m-stein, 2018-05-27:
@BorisMulder-CSL On my sculpt_next branch there's a commit series for the NIC router and the IPXE NIC driver that should fix the problems you described initially in this issue. However, the commits are not yet "official" as they are not merged to master and only testet with my Sculpt scenario by now.

We were able to observe the leaked packets with the IPXE NIC driver but not with the Intel Wifi driver. On link down, the NIC router as well as the IPXE NIC driver do now drop all their buffered packets, then, while down, they do drop all packets that shall be sent. The NIC router does not send back an ICMP error for such packets but informs about the decision whith ``verbose="yes"``.

closed 2018-05-31 by chelmuth