genodelabs/genode #2819: check_port_source for file download may download whole file

chelmuth, 2018-05-09:
As with the now obsoleted nim-csources the jdk port uses a file download from GitHub, which ends in the following command line in check_port_source.
```
 curl -f -L -k -r -2 --max-time 15 --retry 2 https://api.github.com/repos/dmlloyd/openjdk/tarball/64892f1c9874938dfcdc235b5c051dd761cdd833
```
Unfortunately, GitHub does not support the range header and just serves the whole file which takes its time and may result in a timeout.

chelmuth, 2018-09-03:
Unfortunately, the JDK port #2791 again breaks our port checker. It seems GitHub's web servers now fall into the following category (from the curl man page).

> NOTE:  The  file  size is not always known prior to download, and for such files this option has no effect even if the file transfer ends up being larger than this given limit. This concerns both FTP and HTTP transfers.

chelmuth, 2019-02-27:
@nfeske noted that we may change our HTTP download availability test to inspection of the response headers (e.g., status code and `Content-Type`) of the final download location as `-max-file-size` and also `Range` seem to be a compatibility dead-end.

nfeske, 2021-02-26:
There is no work planned for addressing this issue.

commit 3fa1b9ef (2018-05-09): check_port_source: download first kilobytes of file

Unfortunately, some sites do not support the HTTP/1.1 range header and
just serve the whole file, which takes it's time and may result in a
timeout.

Fixes #2819
closed 2021-02-26 by nfeske