genodelabs/genode #2822: hw: add FPU support on ARM* and unify compiler flags

skalk, 2018-05-14:
Recently, we figured out a problem when using special optimized compiler flags for Cortex-A15 during compilation of the Linux IP stack. Hence, we unified the compiler flags of our ARMv7a platforms. As we like to use the same ARMv7a user-land components for different kernels and boards anyway, this step was overdue.

When doing so, I've identified that we do not use the hardware FPU switches for any of our ARM platforms anymore. Therefore, the FPU lazy context-switching in the hw kernel, and the optimized Genode::memcpy are unused, untested code-snippets in our framework. To change that, I've tried to re-enable that code, and unify it across all ARM platforms so that we keep the components platform-agnostic. I've ported an IEEE794 test to stress the implementation. Thereby, I made the following findings:

* Lazy-FPU switching requires compiling the HW kernel without FPU usage, because there is no exception handling for the kernel context
* It is difficult to compile HW's kernel/core solely without FPU compiler flags due to static libraries like "cxx" and "base-common" shared in between core and all other components (and kernels)
* Side-channel attacks and speculative execution in combination with the FPU coprocessor speak against the FPU lazy context-switch
* The VFPv2 FPU variant used in our single ARMv6 platform - the Raspberry PI 1 - is not fully IEEE794 conform. When compiling the whole spectrum of floating-point operations it comes to "undefined instructions" even with enable FPU. Other kernels add emulation code for that holes of the VFPv2

I successfully tested the IEEE794 test on all ARM platforms apart from RPI1. But when executing the complex "noux_tool_chain_auto" run test it stucks silently on Pandaboard and Arndale. At this point, It is not clear to me why the hardcore FPU test running in parallel in four instances succeeds, whereby the implicit FPU usage of the compiler optimizations leads to a freezed system in a more complex test. Maybe someone as an idea? I will interrupt my work on this topic here, because other topics are more urgent right now, but like to document the status quo.

chelmuth, 2018-05-14:
Thanks for the thorough documentation of the current state and findings!

skalk, 2018-11-15:
I've updated my work on the FPU support. To me it seems that the noux_tool_chain_auto problems on base-hw are not directly connected to the FPU-usage, but are a result of the different timings. As we've observed in the last weeks, base-hw regularily fails to finish that test. Therefore, I do not see the coincidence with the FPU usage, and will be lucky when it enters staging at some point.

skalk, 2018-11-27:
There are unsovled problems with TrustZone and ARM virtualization extensions. It first needs to be incorporated into those world-switches, so I mark this as not-fixed again.

skalk, 2018-11-27:
Ok, I've incorporated FPU switching in both world-switches, and put it into the original commit. it is now commit 30d5f77

commit 9561fdc4 (2018-11-16): hw: enable eager FPU context switch for ARM

* Add an ieee754 FPU test
* Remove simple fpu test

Fix #2822
commit b81a9789 (2018-11-27): hw: enable eager FPU context switch for ARM

* Add an ieee754 FPU test
* Remove simple fpu test

Fix #2822
closed 2018-11-29 by chelmuth